adafruit/circuitpython #2839: terminalio.Terminal - VT100 Command Set Extension -> enable text editor on display

kmatch98, 2020-05-01:
The set of commands that are handled in terminalio.Terminal() are insufficient for some editor software with an attached display.

In particular, the following commands are desired:
- Line;ColumnH - move cursor to line, column
- XX;YYm - set highlighting mode
- M - scroll up one line
- Line1;Line2r - scrolling the region between entry Line 1 and Line 2
- 25h and 25l - cursor highlighting on or off

This additional set of commands will enable a text editor like [robert-hh/Micropython-Editor](https://github.com/robert-hh/Micropython-Editor) to run on attached displays.

tannewt, 2020-05-01:
Commands are handled here: https://github.com/adafruit/circuitpython/blob/master/shared-module/terminalio/Terminal.c#L48

I'm pessimistic that we'll have space in the small builds for the extra commands. An alternative would be to implement a second VT100 terminal class in Python that can be loaded as needed.

I only added what was required for the REPL in terminalio.Terminal.

kmatch98, 2020-05-01:
Thanks tannewt, I can definitely understand wanting to keep things small. I’ll find a way to build a separate wrapper for to handle these extra commands.

closed 2020-05-01 by kmatch98

kmatch98, 2020-05-12:
Just a follow up on the Terminal in CircuitPython. I’ve got a simple terminal working that is a wrapper around the tileGrid. It handles a few of the basic functions including showing cursor location, text entry, scrolling.  It’s adequate for a minimalist text editor built from this editor [https://github.com/robert-hh/Micropython-Editor](url)

This library is currently in flux and doesn’t yet meet CP group’s guidelines but perhaps it can point in the right direction. 
[https://github.com/kmatch98/simpleTerminal](url)