adafruit/circuitpython #2840: USB fails to enumerate on nRF52833 with WFI patch

xobs, 2020-05-01:
Beginning with b3b6a64ed7fb10ffb61bc1b71ba4083bca3ada30, USB no longer enumerates when using Simmel, based on the nRF52833.  Reverting that commit fixes USB.

After doing a `mass_erase` of the internal flash and loading the SD and bootloader, I load `firmware.uf2` via USB.  The device resets itself, attempts enumeration, and a short time later triggers an assert:

```
(gdb) c
Continuing.
nrf52.cpu: external reset detected
nrf52.cpu: external reset detected

Program received signal SIGTRAP, Trace/breakpoint trap.
process_set_config (cfg_num=1 '\001', rhport=0 '\000') at ../../lib/tinyusb/src/device/usbd.c:744
744         TU_ASSERT( drv_id < USBD_CLASS_DRIVER_COUNT && drv_len >= sizeof(tusb_desc_interface_t) );
(gdb)
```

dhalbert, 2020-05-01:
Just to double check, did you do a `git submodule update` after pulling the latest? I'd guess so, but I just want to make sure you got the very recent TinyUSB fix that allowed us to proceed with that PR.

I tested this on an nRF52840 on Ubuntu 20.04.

Tagging @hathach, who may want to know more.

xobs, 2020-05-01:
Yes, the tinyusb module is at a74a823b0a140667a4e4e80fa2678870faba2bd0 committed 30/04 at midnight, although the only updates I see to nrf module in tinyusb in the past month have to do with renaming the irq handler to dcd_init_handler.

dhalbert, 2020-05-01:
On what kind of host is this, and do you see the same thing on a different OS?

xobs, 2020-05-01:
That's a very good question.  This is on a Windows machine.

It's difficult to move the board between machines, because the power cables are a little fragile.  However, I also see errors when I plug it into a Raspberry Pi:

```
[539218.392350] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[539218.526172] usb 1-1.4: New USB device found, idVendor=1209, idProduct=c051, bcdDevice= 1.00
[539218.526187] usb 1-1.4: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[539218.526197] usb 1-1.4: Product: Simmel
[539218.526206] usb 1-1.4: Manufacturer: Betrusted
[539218.526216] usb 1-1.4: SerialNumber: 4E84A16AFEBF1E06
[539218.527055] usb 1-1.4: can't set config #1, error -32
```

Windows is much pickier when it comes to USB errors, so I could definitely see it issuing a reset when it failed to set the config, which is probably causing https://github.com/hathach/tinyusb/issues/179

So it does seem like I'm getting USB errors on Linux as well as on Windows, they're just more catastrophic on Windows.

dhalbert, 2020-05-01:
I tried this with a Circuit Playground Bluefruit (nRF52840) on a Dell Intel laptop and a desktop, both running Windows 10, and they seem to be enumerating fine. Both MSC and CDC are working.

I then also tried on an RPi, and the logs don't show the errors you have above.

I'm not sure why it would be peculiar to the '833.

xobs, 2020-05-01:
Me neither.  Maybe it has something to do with v7.0.1 of the SR?  I'm still unclear how that factors into it.

xobs, 2020-05-01:
I just tried on a NRF52833-DK EVK board, and I see the same thing.  If you have one of these, I can give you instructions on how to reproduce the issue.

This is using the Simmel bootloader (https://github.com/simmel-project/bootloader) which shrinks the bootloader partition to 16 kB and removes bluetooth/serial from the bootloader.

dhalbert, 2020-05-01:
Do you have an nRF52840 board you can try with the same CircuitPython commit? It would have 6.1.1, unless you went to some effort to upgrade the build. I guess you could try rolling the '833 SD back to 6.1.1, since it appeared to work. I agree that I don't see why that should make a difference.

I successfully tried with a Feather nRF52840 on Windows too, because the CPB has only a 32MHz crystal. The Feather has both 32kHz and 32MHz, which is more typical.

xobs, 2020-05-01:
@tannewt said he had one, which is why he said I should create a PR for Simmel.

I won't have access to an NRF52840 (specifically a PCA10059) for a few days.

hathach, 2020-05-02:
I have an 833 DK, but I don't see why it could run on 840 but 833. @xobs could you try to flash our stock bootloader on your 833 (best  to be DK)  with SD 6.1.1 to see if that would works. Since we don't enable SD, it wouldn't be an issue for testing USB with it. 

Also it is helpful if you could provide bus capture along with tinyusb log as well by changing `CFG_TUSB_DEBUG` to 2 and retarget using `tu_printf`. It is a bit manual now. 

@dhalbert do we have any printf debug log via RTT SWD now, I can submit an RTT to make getting log from tinyusb easier for future troubleshooting https://github.com/hathach/tinyusb/issues/367

xobs, 2020-05-02:
I created a PR to circuitpython to add the PCA10100 (aka NRF52833-DK).  It uses v6.1.0 You can build it with:

```
make BOARD=pca10100 V=1 SOFTDEV_VERSION=6.1.0
```

It also exhibits the same issue.

xobs, 2020-05-02:
PR is at https://github.com/adafruit/circuitpython/pull/2844

xobs, 2020-05-04:
This is the output I get from RTT:

```
00> Adafruit CircuitPython 5.3.0-128-ge3c6d0f-dirty on 2020-05-04; PCA10100 nRF52833 Dongle with nRF52833
00> USBD init
00> 
00> CDC init
00> MSC init
00> MIDI init
00> code.py output:
00> USBD Bus Reset
00> Hello World!
00> USBD Bus Reset
00> USBD Setup Received 80 06 00 01 00 00 40 00 
00> 
00>   Get Descriptor Device
00> 
00>   Queue EP 80 with 18 bytes
00> USBD Xfer Complete on EP 80 with 18 bytes
00>   Queue EP 00 with zlp Status
00> USBD Xfer Complete on EP 00 with 0 bytes
00> USBD Bus Reset
00> USBD Setup Received 80 06 00 01 00 00 12 00 
00> 
00>   Get Descriptor Device
00> 
00>   Queue EP 80 with 18 bytes
00> USBD Xfer Complete on EP 80 with 18 bytes
00>   Queue EP 00 with zlp Status
00> USBD Xfer Complete on EP 00 with 0 bytes
00> USBD Setup Received 80 06 00 02 00 00 FF 00 
00> 
00>   Get Descriptor Configuration[0]
00>   Queue EP 80 with 64 bytes
00> USBD Xfer Complete on EP 80 with 64 bytes
00>   Queue EP 80 with 64 bytes
00> USBD Xfer Complete on EP 80 with 64 bytes
00>   Queue EP 80 with 64 bytes
00> USBD Xfer Complete on EP 80 with 64 bytes
00>   Queue EP 80 with 26 bytes
00> USBD Xfer Complete on EP 80 with 26 bytes
00>   Queue EP 00 with zlp Status
00> USBD Xfer Complete on EP 00 with 0 bytes
00> USBD Setup Received 80 06 01 03 09 04 FF 00 
00> 
00>   Get Descriptor String[1]
00>   Queue EP 80 with 34 bytes
00> USBD Xfer Complete on EP 80 with 34 bytes
00>   Queue EP 00 with zlp Status
00> USBD Xfer Complete on EP 00 with 0 bytes
00> USBD Setup Received 80 06 00 03 00 00 FF 00 
00> 
00>   Get Descriptor String[0]
00>   Queue EP 80 with 4 bytes
00> USBD Xfer Complete on EP 80 with 4 bytes
00>   Queue EP 00 with zlp Status
00> USBD Xfer Complete on EP 00 with 0 bytes
00> USBD Setup Received 80 06 03 03 09 04 FF 00 
00> 
00>   Get Descriptor String[3]
00>   Queue EP 80 with 18 bytes
00> USBD Xfer Complete on EP 80 with 1800 
00>   Open EP 03 with Size = 64
00>   Queue EP 03 with 31 bytes ... OK
00> 
00>   MSC opened
00> process_set_config 744: assert failed
```

And a screenshot of the USB log:

[image]

It trips the assert fairly quickly, the final SETUP error is due to the assert being triggered which prevents the system from responding.

dhalbert, 2020-05-04:
If you could upload the datacenter packet log as well, that would be helpful to @hathach. Thanks!

I have an nRF52833 DK board arriving in a day or two, and will try to reproduce the problem.

xobs, 2020-05-04:
[nrf52833-dk-failure.zip](https://github.com/adafruit/circuitpython/files/4572618/nrf52833-dk-failure.zip)

xobs, 2020-05-04:
I'm beginning to think it's not hardware related, but rather to the fact that, in order to save space, I have `CIRCUITPY_USB_HID = 0`.  I still see the HID descriptors in the config descriptor that gets sent out.

I think the nRF52833 is a red herring.  Setting `CIRCUITPY_USB_HID = 1` fixes the issue.

I'll update the title of this bug.

tannewt, 2020-05-04:
Closing this in favor of #2850 now that we understand the issue.

closed 2020-05-04 by tannewt