adafruit/circuitpython #2846: MemoryError on restart after copying new code.py for application which has plenty free

kevinjwalters, 2020-05-03:
I can't reproduce this on demand but I just noticed I got a `MemoryError` after making minimal changes to a progam and copying `code.py` over an old version to re-deploy it to an Adafruit CLUE running `5.2.0`. I only had the TFT LCD screen active and it mentioned `12000` bytes. The code actually has a forced GC in the main loop which has been there for some time. I added a debug print of the `mem_free()` to see if I'd accidentally got close to running out of memory and I'm nowhere near it, over 50k free.

```
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
mem_free=55552
0 976.032 1503.79 0.261612 1503.78 1.5029 None
mem_free=55456
1 976.186 1503.21 0.904701 1503.38 1.5029 None
mem_free=55456
2 976.275 1503.52 0.670845 1503.43 1.5029 None
mem_free=55456
3 976.388 1502.55 0.524689 1502.84 1.5029 None
mem_free=55456
4 976.477 1502.9 0.626999 1502.79 1.5029 None
mem_free=55472
5 976.604 1503.03 0.685463 1502.99 1.5029 None
mem_free=55472
6 976.69 1502.4 0.510071 1502.59 1.5029 None
mem_free=55472
7 976.803 1502.56 0.410713 1502.51 1.5029 None
mem_free=55472
8 976.891 1503.58 0.787773 1503.27 1.5029 None
mem_free=55456
9 977.02 1502.43 0.00147247 1502.77 1.5029 None
mem_free=55456
10 977.109 1503.18 0.407768 1502.95 1.5029 None
mem_free=55456
11 977.219 1502.67 0.18853 1502.82 1.5029 None
mem_free=55456
12 977.308 1501.94 0.933929 1502.16 1.5029 None
mem_free=55440
13 977.439 1502.01 0.49546 1501.98 1.5029 None
mem_free=55440
14 977.528 1503.27 0.0131454 1502.89 1.5029 None
mem_free=55440
15 977.64 1502.01 0.407768 1502.39 1.5029 None
mem_free=55440
16 977.727 1502.39 0.948547 1502.27 1.5029 None
mem_free=55440
Traceback (most recent call last):
  File "code.py", line 563, in <module>
  File "code.py", line 415, in sample_sum
  File "code.py", line 415, in <listcomp>
KeyboardInterrupt:

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.2.0 on 2020-04-09; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
```

I have seen things like this before but I think I was closer to the memory limit so was less surprised. A previous discussion on discord from 18-Mar-2020:

> **kjw** I'm getting pretty close to the memory limit of the CLUE (running 5.0.0 with some CircuitPython. I've noticed sometimes when I copy an updated code.py onto it I get a MemoryError but then it's fine on reloads after that. Anyone else seen this?

> *danh* : that is odd; there are few things that stay in RAM between soft reloads, but I don't know why a second soft reload would fix things. There is a 4k filesystem buffer that gets created on demand on filesystem writes, if I remember correctly. It certainly sounds like you are at the hairy edge.

kevinjwalters, 2020-06-12:
This one is odd and potentionally along the same lines. I did a very minor tidy and copied the code across to a CLUE's CIRCUITPY and it blew up with a `MemoryError`:

```
NO collect mem_free  17072
Traceback (most recent call last):
  File "code.py", line 962, in <module>
MemoryError: memory allocation failed, allocating 2048 bytes

Code done running. Waiting for reload.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Very conveniently and coincidentally (?) that was just before the only place in the code where I print `mem_free()`

```
$ cat clue-multi-rpsgame/code.py | cat -n | sed -ne '958,962p'
   958          if debug >= 2:
   959              d_print(2, "NO collect mem_free ", gc.mem_free())
   960
   961          ### This sound cue is really for other players
   962          audio_out.play(WaveFile(audio_files["ready"]))
```

There's no `gc.collect()` before this or anywhere in the code at the moment. I've been using this code for sometime without any memory issues on three boards. Either I have just got close to the limit or there's something odd going on with this case, I think it's the latter.

deshipu, 2020-06-13:
Have you tried pre-allocating the buffer for audioio, so that it can reuse the same part of memory?

kevinjwalters, 2020-06-13:
@deshipu Not yet, but allocating that early and once is a good idea, thanks. The feature looks like part of 5.x [WaveFile](https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/audiocore/WaveFile.html).

> `buffer` (bytearray) – Optional pre-allocated buffer, that will be split in half and used for double-buffering of the data. If not provided, two 512 byte buffers are allocated internally.

That mentions 2x512 which doesn't match 2048. I wonder if it scales to the sample_rate.

kevinjwalters, 2020-06-13:
Hmm, I switched to this

```
gc.collect()
file_buf = bytearray(2048)

audio_out.play(WaveFile(audio_files["welcome-to"], file_buf))
```

And I now have three boards all that have bizarre errors including one in the `HardFault_Handler`. I think something is crapping all over memory.

dhalbert, 2023-10-28:
Closing this because we'd like to see the same symptoms now in a reproducible way. Many things have changed, including with nRF impls since this report.

closed 2023-10-28 by dhalbert