adafruit/circuitpython #2848: Add vertical scrolling to TileGrid

kmatch98, 2020-05-04:
Per @tannewt there is currently vertical TileGrid scrolling available in the C libraries. Access to the underlying library would likely speed up vertical scrolling of a TileGrid. 

This is a request to allow python code access to the available underlying TileGrid C code for scrolling.

tannewt, 2020-05-04:
This is what I was thinking of for the "scroll": https://github.com/adafruit/circuitpython/blob/master/shared-module/displayio/TileGrid.c#L307

Not sure what a good Python level API would be.

kmatch98, 2020-05-05:
I’m no expert, but if you open up being able to change which tile is at the origin, then any other types of scrolling could be added as wrappers around it. 

Currently scrolling requires rearranging the tilegrid, which tends to be rather slow. If you’re mostly scrolling in one direction, then is actually faster to move a bunch of smaller tilegrids and just update the x,y locations of each. 

So anything you can give access out of the C code will likely give performance benefits. 

Maybe @deshipu would have more insight.

tannewt, 2020-05-05:
Two other hang ups I have about exposing this are:
1) it causes the whole tile grid to refresh. Disabling autorefresh and resetting all of the indices is equivalent.
2) It'd be better to scroll at a higher level because SPI displays have a native scroll in one direction that would be much faster to use.

kmatch98, 2020-05-05:
I misunderstood that my main performance issue was due to the autorefresh that redrew the tilegrid after each change.  I changed my code to turn autorefresh off before re-arranging all the tilegrids, followed by turning autorefresh on again.  This made a huge improvement.  

I'll close this out as a misunderstanding on my part about the performance limiting factor.  For the scrolling I was doing, the limiting factor was the auto refresh, not the redefinition of the tilegrids.

commit aa34c553 (2017-09-28): py/nlr: Fix execstack builds for ARM.

It seems that the gcc toolchain on the RaspberryPi
likes %progbits instead of @progbits. I verified that
%progbits also works under x86, so this should
fix #2848 and fix #2842

I verified that unix and mpy-cross both compile
on my RaspberryPi and on my x64 machine.
closed 2020-05-05 by kmatch98