adafruit/circuitpython #2852: _bleio: Pass connection to PacketBuffer, its Characteristic, or its Service

dhalbert, 2020-05-04:
`PacketBuffer` needs to know about its `Connection` to determine `.incoming_packet_length` and `.outgoing_packet_length`. When its `Characteristic` is remote, the `Connection` is know via the `Service`, but when the `Characteristic` and `Service` are local, the `Connection` is not easily found. Right now, `PacketBuffer` grabs the `conn_handle` by spying on GATTS write operations, and also notices disconnections.

It might make more sense for the Python program simply to pass the `Connection` in, once a connection has been established. This could be to the `PacketBuffer` itself, or to the `Service`. An API like `Service.connection =` or similar could be used.

This would be an incompatible change, deferred until 6.0.0.

tannewt, 2020-07-14:
@dhalbert Still want to do this in 6.0.0 or punt to 7.0.0?

dhalbert, 2020-07-14:
Ok to move to 7.0.0. I'll change the milestone.