adafruit/circuitpython #2853: _bleio: remove PacketBuffer.packet_size for 6.0.0

dhalbert, 2020-05-04:
In 5.x, after #2799 is merged, `PacketBuffer.packet_size` is a synonym for `PacketBuffer.incoming_packet_length`, and is deprecated.

In 6.0.0, it can be dropped, after updating the libraries that use it.

dhalbert, 2020-06-28:
`.incoming_packet_length` was added after 5.3.0, so unless we have another 5.x stable release, and then change the libraries, we can't drop `.packet_size` and have the libraries be backwards compatible without conditional code. We could defer this to 7.0.0, or put in conditional code.

The Adafruit libraries that use `.packet_size` now are:
https://github.com/adafruit/Adafruit_CircuitPython_BLE_Adafruit
https://github.com/adafruit/Adafruit_CircuitPython_BLE_iBBQ
https://github.com/adafruit/Adafruit_CircuitPython_BLE_Heart_Rate
https://github.com/adafruit/Adafruit_CircuitPython_BLE_MIDI

tannewt, 2020-06-30:
I'm fine waiting for 7.0.0.

commit 5ec195b5 (2021-04-08): Remove PacketBuffer.packet_size

Fixes #2853
closed 2021-04-08 by dhalbert