adafruit/circuitpython #2854: Turn off CircuitPython branding by default

tannewt, 2020-05-04:
Builds should default to a different name instead of CircuitPython. That way forks and their builds (including dev builds) are not branded as CircuitPython. We’d have an environment variable to specify it and set it in our CI. No one else should set it to CircuitPython (but other names are totally cool.) Name ideas (bonus points for it being a Monty Python reference):

- VoltageEel
- ElectricEel
- ElectricSnake
- PhysicEel
- EelTronics
- EmbeddedEel
- Eels on a PCB
- Lumberjack
- Blancmange
- CircuitShrubbery
- FlyingCircusPython
- NiPython

We also want to make sure that when falling into the default name a notice is also provided over serial:

```Python for microcontrollers based on CircuitPython 5.2.0 on 2020-04-09; Adafruit Grand Central M4 Express with samd51p20
Note: This is an unofficial build of CircuitPython. This build is not supported and may not be distributed as an official CircuitPython build, as it may break compatibility with libraries and tutorials
See {short_url} for more details.```

Credit to @hierophect, @theaflowers, @k0d, @dhalbert for the wording. Do we actually want a name or just leave it as "Python for microcontrollers"?

To do this we'll need to find all references that folks see and choose which to change. (CIRCUITPY should probably stay the same, for example.)

hierophect, 2020-05-05:
Here's the fully workshopped version:

```
Python for microcontrollers based on CircuitPython 5.2.0 on 2020-04-09; Adafruit Grand
Central M4 Express with samd51p20
Note: This is an unofficial build of CircuitPython. This build is not supported and may not be
distributed as an official CircuitPython build, as it may break compatibility with libraries and
tutorials
See {short_url} for more details.
```

hierophect, 2020-05-05:
if we were to do it with a name I'd do something like this:
```
NiPython for microcontrollers based on CircuitPython 5.2.0 on 2020-04-09; Adafruit
Grand Central M4 Express with samd51p20
Note: NiPython is our name for unofficial builds of CircuitPython. These builds are
not supported and may not be distributed as official CircuitPython builds, as they
may break compatibility with libraries and tutorials
See {short_url} for more details.
```

k0d, 2020-05-05:
I really feel that a new name would be very confusing and is not needed. Python for microcontrollers sounds right, it's not a new name...just a way of explaining what it is and what it is for. The 'workshopped version' that @hierophect wrote above, that's what I'd vote for.

dhalbert, 2020-05-05:
I agree about avoiding a new name.

BTW, it turns out "NI Python" is used by National Instruments.

geekguy-wy, 2020-05-05:
The phrase "Python for microcontollers" might cause confusion. That phrase is used several times to describe Micropython.

hierophect, 2020-05-05:
> The phrase "Python for microcontollers" might cause confusion. That phrase is used several times to describe Micropython.

I don't see that as a big issue, it's used persistently in the Circuitpython docs as well. And I think it actually plays into the point, that without being official builds are just another subversion of the Micropython base concept but without any guarantees attached.

hugodahl, 2021-04-14:
Since the original discussion goes back nearly a year, I wanted to confirm what we're expecting.
From what I understand, we want to replace _at least_ the following instances of "CircuitPython" and "Adafruit CircuitPython" with "Python for microcontrollers":
* [ ] REPL welcome line
* [ ] The banner line of the `help()` command
* [ ] The text written to "boot_out.txt"
* [ ] Error or notification messages, such as the `print_safe_mode_message()` method in [safe_mode.c](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/safe_mode.c#L122), which is how *I* got here

However, just above that method, we have this line:
```c
#define FILE_AN_ISSUE translate("\nPlease file an issue with the contents of your CIRCUITPY drive at \nhttps://github.com/adafruit/circuitpython/issues\n")
```

How should something like this be addressed?

Something else which I think might need to be addressed, either as part of this issue or a separate one, once this issue sets the pattern, are other pieces of information. For example, looking at calling `help()` from the REPL

```text
>>> help()
Welcome to Adafruit CircuitPython 6.2.0-beta.1!

Please visit learn.adafruit.com/category/circuitpython for project guides.

To list built-in modules please do `help("modules")`.
```

Would we want to allow for the second line, the "Please visit..." to be adjustable as well? Let's say SparkFun create their own branded build of Python for microcontrollers - I like to imagine they'd call it "SparkyPy" - they may want the `help()` output to point to their learning system, most likely with some other verbiage around it.

hugodahl, 2021-04-14:
Another tricky instance we're likely to run into are DocStrings. For example, in the [`/shared-bindings/supervisor/RunReason.c`](https://github.com/adafruit/circuitpython/blob/main/shared-bindings/supervisor/RunReason.c#L36-L50).

In this situation, it's a simple case of replacing with the correct value, however, other DocString instances may be more involved. More of a heads up as something to keep in mind.

dhalbert, 2021-04-14:
If we need to conditionalize the doc strings in the special `//|` comments, I found the `cpp` `-C` and `-CC` flags, which pass through comments instead of discarding them. However, if a comment is inside a false `#if` or similar, it will be discarded. So there is a mechanism to conditionalize the docstrings.

Or, since we are extracting the docstrings and processing them as Python, maybe there's a Python-oriented way of doing it, by doing computations to build the docstrings or select among them.

deshipu, 2021-04-14:
Not sure if this belongs here, but I would also love the option of removing the Blinka logo from the REPL and reclaiming the extra columns that uses.

tannewt, 2021-04-14:
@hugodahl I think we can do this in pieces. The four items you have on the TODO make sense on where to start. Docs can be done in a follow up. I think the blinka icon in the console can be separate as well.

I do like the idea of configuring the issue location as well because this is about support. We want to make it clear what builds of CP are supported by CP official.

hugodahl, 2021-04-19:
Thanks for the feedback everyone. As Scott mentioned, I'll start with the TODOs I've outlined above, and go from there.

For the time being, I've created a Gist with what I've come up with to define the replacement strings, as well as (pseudo) implementation in Python for the process. 

Gist: https://gist.github.com/hugodahl/67c01a8c473804b9c148a0e4b0b1a8b5

dhalbert, 2021-04-21:
Currently there is a `USB_INTERFACE_NAME` make macro that is normally set to `"CircuitPython"`. But a few boards change this, currently to either `"Sol"` or `"StringCarM0Ex"`.

`USB_INTERFACE_NAME` is currently used as a prefix when constructing USB interface names, so we have, for instance, "CircuitPython CDC control". I just started using this prefix in https://github.com/adafruit/Adafruit_Board_Toolkit/blob/main/adafruit_board_toolkit/circuitpython_serial.py to identify CircuitPython boards and to help identify their REPL and data USB CDC channels. The Mu Editor is going to depend on `Adafruit_Board_Toolkit` to do these identifications.

I am currently changing `Adafruit_Board_Toolkit` to handle these unusual boards. Going forward, we can change the CDC interface names to something more general, or match a pattern instead of just checking a prefix. Or, we can just always use "CircuitPython" in the CDC prefixes. (There's utility in using a different prefix in other cases: "Sol" is used to help distinguish MIDI interfaces, for instance.). Of course `adafruit_board_toolkit` would be backward compatible.

This is assuming we want unbranded boards to be usable from Mu in "CircuitPython" mode. The consequences of not doing that are painful to the end user, though we can't guarantee the boards would behave like CircuitPython in the editor.

tannewt, 2021-04-21:
I'm ok leaving the CircuitPython name in "APIs" so tools continue to work. I don't mean to block interoperability.