adafruit/circuitpython #2855: NRF: USB Disconnect / Reconnect / Enumeration fails during sleep

xobs, 2020-05-05:
With the most recent patches, issuing `time.sleep(x)` issues a `wfi` instruction.  This aggressively puts the CPU to sleep.  However, this appears to cause USB to no longer enumerate.  While this is not an issue with a device that's powered off of USB, if a device has its own battery, then this will prevent the device from enumerating during sleep.

## Steps to reproduce

The following `code.py` should trigger this bug:

```python
from time import sleep
sleep(60)
```

xobs, 2020-05-05:
I believe at least part of the problem is due to the fact that there doesn't seem to be anything listening to the `USBDETECTED` and `USBREMOVED` events.  This patch adds that in:

```patch
diff --git a/ports/nrf/supervisor/port.c b/ports/nrf/supervisor/port.c
index 539a9cf90..24a0c4bdd 100644
--- a/ports/nrf/supervisor/port.c
+++ b/ports/nrf/supervisor/port.c
@@ -67,6 +67,10 @@ static void power_warning_handler(void) {

 const nrfx_rtc_t rtc_instance = NRFX_RTC_INSTANCE(2);

+#if NRF_POWER_HAS_USBREG
+extern void tusb_hal_nrf_power_event(uint32_t event);
+#endif
+
 const nrfx_rtc_config_t rtc_config = {
     .prescaler = RTC_FREQ_TO_PRESCALER(0x8000),
     .reliable = 0,
@@ -115,6 +119,12 @@ safe_mode_t port_init(void) {
     nrfx_power_pof_init(&power_failure_config);
     nrfx_power_pof_enable(&power_failure_config);

+    #if NRF_POWER_HAS_USBREG
+    const nrfx_power_usbevt_config_t usbevt_config = { .handler = ((nrfx_power_usb_event_handler_t) tusb_hal_nrf_power_event) };
+    nrfx_power_usbevt_init(&usbevt_config);
+    nrfx_power_usbevt_enable();
+    #endif
+
     nrf_peripherals_enable_cache();

     // Configure millisecond timer initialization.
```

That at least gets it part of the way there.  However, it still appears as though the device isn't responding to USB properly.  In particular, while the config descriptor and initial string descriptor get transferred, the device doesn't appear to be acknowledging them.
[enumerate-with-wfi.zip](https://github.com/adafruit/circuitpython/files/4580516/enumerate-with-wfi.zip)

xobs, 2020-05-05:
[image]

tannewt, 2020-05-05:
@hathach Does tinyusb use interrupts for these USB events? Sleep will wake and call `tud_task` after any interrupt.

hathach, 2020-05-05:
TinyUSB requires application to call `tusb_hal_nrf_power_event()` within the power handler. 
https://github.com/hathach/tinyusb/blob/master/src/portable/nordic/nrf5x/dcd_nrf5x.c#L630

The power IRQHandler is used and shared with SoftDevice therefore it cannot be actively handled when SD is enabled. The above modification by @xobs is part of the initial `init_usb_hardware()` when enabling USB (with SD not enabled yet). I am not sure how WFI() affect the power peripheral, but in case it does, re-init is probably needed.

xobs, 2020-05-06:
I thought it was odd that it seemed to somewhat work even without that patch.

The strange part is I can see the events fire, and I see the `tusb_hal_nrf_power_event()` function fire with all three kinds of events, but it still just does not work.  Do we need to be calling `tusb_task()` more often now?  Should `WFI` be disabled when not connected to `VUSB`?

tannewt, 2020-05-06:
> I thought it was odd that it seemed to somewhat work even without that patch.
>
> The strange part is I can see the events fire, and I see the `tusb_hal_nrf_power_event()` function fire with all three [...]

My guess is that there is an interrupt we need to enable so that it wakes up from WFI. Did you confirm that it works when the actual WFI call is commented out? This looks very similar to the SAMD21 issue where a SETUP packet interrupt is piggybacking on a OUT interrupt. It still failed when WFI was commented out because it was a timing issue.

> Should `WFI` be disabled when not connected to `VUSB`?

Definitely not. It's only really useful when on battery. When on USB you have an external power supply.

xobs, 2020-05-07:
It does work if the WFI call is commented out.

Let me look through the USB interrupts and see what's going on.

xobs, 2020-05-07:
@hathach You're right, I didn't see that the event was already being hooked.  So the patch I made probably doesn't seem necessary.

However, it does appear as though the OUT data isn't getting ACKed for whatever reason.  And it's always the same descriptor.  At least it's consistent.

The contents of the `USBD_INTEN` register prior to unplugging:
```
(gdb) x 0x40027300
0x40027300:     0x0185d57d
(gdb)
```

And after plugging back in, while the host is waiting for the device to ACK the 5th SETUP packet:
```
(gdb) x 0x40027300
0x40027300:     0x01801405
(gdb)
```

xobs, 2020-05-07:
When I connect it when it's in this state, this is the output I receive:

```
USBD Unplugged

USBD Bus Reset
USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset
USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 02 00 00 FF 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 26 bytes
USBD Xfer Complete on EP 80 with 26 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 01 03 09 04 FF 00
  Get Descriptor String[1]
  Queue EP 80 with 34 bytes
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 03 00 00 FF 00
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes

USBD Setup Received 80 06 00 02 00 00 DA 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes
USBD Bus Reset
USBD Setup Received 80 06 00 01 00 00 40 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset
USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 01 00 00 12 00
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 02 00 00 09 00
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with zlp Status
[... 86 lines cut ...]
USBD Setup Received 80 06 01 03 09 04 02 00
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes
USBD Setup Received 80 06 01 03 09 04 02 00
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 01 03 09 04 22 00
  Get Descriptor String[1]
  Queue EP 80 with 34 bytes
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received A1 FE 00 00 02 00 01 00
  MSC control request
  Queue EP 80 with 1 bytes
```

I have placed multiple "\n" entries where the host is waiting for the device to ACK an IN packet.

Note that this board is being debugged by bit-banging SWD on a raspberry pi, so I needed to hack in rtt support.  I think this is accurate, though.

Also, my USB Beagle is in need of a new inductor.  It is incredibly flaky right now, making it hard to get reliable measurements, particularly when it involves a lot of unplugging and plugging in.

xobs, 2020-05-08:
In looking at https://github.com/hathach/tinyusb/blob/master/src/portable/nordic/nrf5x/dcd_nrf5x.c#L292-L297 it almost seems as though this isn't actually firing.

One thing I can see is that the `_usbd_q` is getting written to but not read from.  If I break into the debugger while it's doing `wfi`, I see that wr_idx != rd_idx:

```
(gdb) p *_usbd_q
$10 = {
  role = 1 '\001',
  ff = {
    buffer = 0x2000b740 <_usbd_qdef_buf> "",
    depth = 16,
    item_size = 12,
    overwritable = false,
    count = 1,
    wr_idx = 7,
    rd_idx = 6
  }
}
(gdb)
```

xobs, 2020-05-08:
The issue is that interrupts were still enabled while calling `wfi`.

This PR adds a patch to disable interrupts (and run background tasks with interrupts disabled to ensure buffers are drained) on the nrf platform: https://github.com/adafruit/circuitpython/pull/2868

I suspect other platforms will have similar subtle issues.

xobs, 2020-05-08:
Looks like it's not possible to run `tud_task()` with interrupts disabled.

@hathach do you have any suggestions for how we could do this?

xobs, 2020-05-08:
I've applied this patch to my local copy of tinyusb, which causes it to poll the interrupt handler if interrupts are disabled.  This seems to prevent the deadlock:

```patch
diff --git a/src/portable/nordic/nrf5x/dcd_nrf5x.c b/src/portable/nordic/nrf5x/dcd_nrf5x.c
index 12435ea2..acc1af58 100644
--- a/src/portable/nordic/nrf5x/dcd_nrf5x.c
+++ b/src/portable/nordic/nrf5x/dcd_nrf5x.c
@@ -101,7 +101,11 @@ static void edpt_dma_start(volatile uint32_t* reg_startep)
     else
     {
       // Otherwise simply block wait
-      while ( _dcd.dma_running )  { }
+      while ( _dcd.dma_running )  {
+        // Manually poll the handler if interrupts are disabled
+        if (!__get_PRIMASK())
+          dcd_int_handler(0);
+      }
     }
   }
```

xobs, 2020-05-08:
I submitted a PR for this here: https://github.com/hathach/tinyusb/pull/396

I've also found additional places where `RUN_BACKGROUND_TASKS` assumes interrupts are enabled, These have been rolled into https://github.com/adafruit/circuitpython/pull/2868

tannewt, 2020-05-08:
Is the cause for this a race between the last background task run and the wfi call? WFI should work fine with interrupts enabled but for TinyUSB that means work may have been queued but not done.

xobs, 2020-05-09:
WFI does work fine if interrupts are enabled, but without disabling interrupts you can never be sure that there isn't pending data before issuing `WFI`  For example:

```asm
sleep_until_time:
       ; ... do some prep involving the RTC
       bl         usb_background
       wfi

usb_background:
       ; ... service USB buffers
       bx lr
```

Even though `usb_background()` drains the buffer, you still could have an interrupt arrive between the `bx lr` and the `wfi` instruction.  If interrupts are enabled, then that will get serviced and there will be data waiting in the queue.  If that is a packet from the host, then you will go to sleep without responding to that packet and the host will time out the device.

The solution is to disable interrupts, flush buffers, and issue `wfi`.  If there's a pending interrupt it will return immediately, allowing you to re-enable interrupts and have it serviced.

commit 77cf4dce (2020-05-21): nrf: disable interrupts before running wfi

In order to ensure we don't have any outstanding requests, disable
interrupts prior to issuing `WFI`.

As part of this process, check to see if there are any pending USB
requests, and only execute the `WFI` if there is no pending data.

This fixes #2855 on NRF.
closed 2020-05-21 by tannewt