adafruit/circuitpython #2856: STM32: implement a cross-module timer allocator

hierophect, 2020-05-05:
The STM32 timer system has multiple different kinds of timers - basic, general purpose, and advanced - which are typically tied to pins or specific peripherals. In comparison to other ports, the timers have somewhat limited multiplexing to pins, with some timers only having a single viable pin and vice versa. 

Currently, to avoid taking pin-timer combinations out of the "pool", multiple modules are competing for the "basic timer" set, which have no pin assignments at all. However, there are only up to two basic timers on a given chip and many "access-line" STM32 chips do not have them at all. This is complicating the inclusion of new modules like RotaryIO and disables popular modules like Pulsio on lower-end SoCs. 

It may be time to implement some form of timer mediator in Microcontroller that handles the allocation of timers across different modules. This mediator would be able to assign general purpose, basic and advanced timers, prioritizing those that are on seldom used pins, or pins that are not even available on most packages.

I don't know if it might also be a good idea to give the user some optionality about what timers get assigned where? It requires more knowledge about ST's specific timer situation, but it isn't that hard to imagine a scenario where someone has a board that needs a very specific pin to have the timer on PWM mode and a "dumb" allocator picks it for use with PulseIO timings anyway...

@tannewt @jepler

tannewt, 2020-05-05:
👍 I'd like to see this because it'll make PulseIn and PulseOut work on all STM boards.

hierophect, 2020-05-05:
It's also relevant for Protomatter and RotaryIO which likewise won't work without it