adafruit/circuitpython #2859: .bin files for uChip release build?

mew-cx, 2020-05-06:
Several weeks ago, #2804 was merged that added uChip .bin files to the [S3 archive](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/uchip/en_US/) with the .uf2 files, and it has been working great for nightly builds.

However, for the recent release of `5.3.0.uf2` there isn't an associated `5.3.0.bin` file in the [S3 archive](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/uchip/en_US/)  I spot-checked a couple other boards using the `BIN_UF2` flag, and they did have both .uf2 and .bin for the 5.3.0 release in S3.

Is there another process to be modified that will copy the .bin files for release builds?

jepler, 2020-05-06:
5.3.0 was released from a point before #2804 was merged, even though chronologically the release was made after the PR was merged. If the next release is made from master branch, then it will be included. If a 5.3.x with this fix is needed, a fresh PR onto the 5.3.x branch to make the same change would be required.

mew-cx, 2020-05-07:
Thanks @jepler that explains it.

closed 2020-05-07 by mew-cx