adafruit/circuitpython #2860: Disable REPL to use USB UART for custom console?

AdinAck, 2020-05-06:
I am using an M4 ItsyBitsy and would like to create a custom console for when I plug my computer into the USB port on the ItsyBitsy. I don't know how to access the Tx and Rx pins on the USB port and I don't know if I would be able to use them even if I did know what pins they were since the REPL uses them. And yes I know that there is input() and print() but I need background processes to occur even while the user is using the console (since input() holds the whole script).

Is there any way to accomplish this?

Thanks!

deshipu, 2020-05-06:
There are on TX and RX pins on the USB port — that serial port is virtual, simulated in software.

If you want non-blocking input, you will probably be interested in issue #231.

tannewt, 2020-05-06:
Closing this because it is a duplicate of #231.

closed 2020-05-06 by tannewt