adafruit/circuitpython #2861: STM32: Feather F405 stuck in port.c

hierophect, 2020-05-06:
Experiences a hang at startup at `while(__HAL_RCC_GET_FLAG(RCC_FLAG_LSERDY) == RESET)`. Likely due to some problem with the LSE.

hierophect, 2020-05-07:
@tannewt requesting assistance. Can you replicate this on your feather?

hierophect, 2020-05-07:
Confirmed issue does not occur on the OpenMV, STM32H7.

jerryneedell, 2020-05-07:
Confirmed that it does not boot with master at c3b3eb4c2
it does boot with 5.3.0.

tannewt, 2020-05-08:
@hierophect I don't have cycles to look into this now. I think you fixed it in another PR already anyway. :-)

hierophect, 2020-05-11:
@tannewt I'm still hanging at boot for the Feather. Just did some additional testing and it seems to also brick the Pyboard, but interestingly, NOT the F411 boards (pyb nano, blackpill). I'll work on tracking this down today if you're busy.

hierophect, 2020-05-11:
@ladyada do you think I could get the mfn part numbers for the LSE and caps used on the Feather?

tannewt, 2020-05-11:
@hierophect still need the part info? I can look it up too.

hierophect, 2020-05-12:
@tannewt I would find it interesting to know for future reference, but clearly it wasn't the cause of the problem, so I don't need it for anything specific. If nothing else had worked I would have run it through STM32's appnote of oscillator compatibility formulas for possible problems.

tannewt, 2020-05-12:
Kk, it's not super easy for me to get so I'll just close this. I believe #2874 fixes this.

closed 2020-05-12 by tannewt

hierophect, 2020-05-12:
@tannewt I think you mean #2866? If so yes that was the intended fix, I forgot to put in the Resolves keyword.