adafruit/circuitpython #2864: Feather STM32F405 Express

Electro-Mike, 2020-05-07:
Any information on being able to use Circuitpython on the STM32F405 to play wav files?

Thanks.
Mike

shalikashvili, 2020-12-07:
I would strongly encourage development of the audioio module, as the STM32F405 is one of the more powerful processors in its class, and this module would facilitate applications that would be infeasible on lesser platforms.

dastels, 2021-02-10:
Agreed, shalikashvili, audio is a pretty common feature for many projects. It's lack on this platform is notable.

tannewt, 2021-02-10:
We'd love to have it but it isn't a priority for us. We're happy to help anyone get started developing with CircuitPython.

iraytrace, 2021-05-01:
So, could we encourage Adafuit to update the description at: https://www.adafruit.com/product/4382 to indicate that there is NOT audio support?

Just out of curiosity, what would it take to get this working?  I'd be willing to discuss the possibility of implementing.

ladyada, 2021-05-01:
the description should not say audio - its been removed.