adafruit/circuitpython #2865: NRF: pressing Ctrl-C breaks `time.sleep()` until reboot

xobs, 2020-05-07:
# Description

Pressing `Ctrl-C` to interrupt `time.sleep()` will cause `time.sleep()` to instantaneously return on NRF until the board is reset.

# Steps to reproduce

In the REPL, run:

```python
import time
time.sleep(100)
```

While it's sleeping, press Ctrl-C.  The REPL will come back.  Then run:

```python
time.sleep(100)
```

The REPL will return instantly.

# Workaround

Performing a soft- or hard-reboot re-initializes the sleep timer and gets it working again.

tannewt, 2020-05-07:
Is this on both nRF52840 and nRF52833? Please let us know what boards you've found this on. Thanks!

dhalbert, 2020-05-07:
I see this on nRF52840.

dhalbert, 2020-05-07:
This is not just nRF. I see the same thing on Metro M0.

dhalbert, 2020-05-07:
This broke somewhere between 3.1.2 and 4.1.2

dhalbert, 2020-05-07:
Commit causing the issue is ea760c042bcdff61792a4b5f3de0b3ec212c44c4: "Turn on MICROPY_OPT_COMPUTED_GOTO for 5x CPU-bound speedup". So probably there's something wrong with how `MP_STATE_VM(mp_pending_exception)` is handled when `MICROPY_OPT_COMPUTED_GOTO` is turned on

dhalbert, 2020-05-08:
This problem does not occur in current micropython even when `MICROPY_OPT_COMPUTED_GOTO = 1`.

I compared our `py/vm.c` with micropython's, because `MICROPY_OPT_COMPUTED_GOTO` is only significant in that file. There are differences in exception handling based on the flag.

Really interestingly, this micropython commit, which is a minor speedup, https://github.com/micropython/micropython/commit/869024dd6e62905b7e1069b547856a769b3b24ba, changes
```
    #define DISPATCH() break
```
to
```
    #define DISPATCH() goto dispatch_loop
```
If I make that same change in our `py/vm.c`, then I can reproduce the problem _even when_ `MICROPY_OPT_COMPUTED_GOTO = 0`. I haven't been able to track down other micropython changes yet that are causing this problem not to occur on micropython, though I will go back a year or two at a time in micropython in ports with computed goto turned on and see what happens. It's fishy in the previous comment that the interrupt doesn't get printed.

dhalbert, 2020-05-08:
Aha: https://github.com/adafruit/circuitpython/issues/1092

kevinjwalters, 2020-05-10:
Is this a dupe of #2691 ?

#2689 features more control-c / interrupt woes.

tannewt, 2020-07-14:
This is fixed by #3043. Thanks @DavePutz!

closed 2020-07-14 by tannewt