adafruit/circuitpython #2874: STM32F405 crashes to hardfault

jerryneedell, 2020-05-10:
I was truing an I2C device with the STM32F405 undere CP 5.3.0 and it crashes to the HardFault Handler
```Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
                                 Press any key to enter the REPL. Use CTRL-D to reload.

``` when I try to run the ht16k33_segments_simpletest

the code runs fine under 5.2.0.

I tried another test using an SPI device and it also went to the Hardfault handler...
is 5.3.0 broken for the STM32F405?

jerryneedell, 2020-05-10:
another piece of information 
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Feather STM32F405 Express with STM32F405RG
>>> import adafruit_msa301
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_msa301.py", line 53, in <module>
  File "adafruit_register/i2c_struct.py", line 41, in <module>
RuntimeError: maximum recursion depth exceeded
>>> 
```

hierophect, 2020-05-11:
Interesting, I haven't had that issue when testing with SPI devices. This issue is currently super-ceded by the #2861 issue on master, but I'll have it next up on my list.

jerryneedell, 2020-05-11:
FYI ```Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Feather STM32F405 Express with STM32F405RG
>>> import adafruit_rfm9x
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_rfm9x.py", line 49, in <module>
  File "adafruit_bus_device/spi_device.py", line 33, in <module>
RuntimeError: maximum recursion depth exceeded
>>> 
```

hierophect, 2020-05-11:
Actually, I realize that the `maximum recursion depth` implies this is likely related to #2830. You could try manually expanding the stack size in the stm32f405_default file to 24k. I've encountered some of my own hardfaults when testing this, though, so let me know if you try it and it works.

jerryneedell, 2020-05-12:
resolved with 5.4.0 release  -- boots normally and tested both I2C and SPI devices

jerryneedell, 2020-05-12:
fixed by #2866

closed 2020-05-12 by jerryneedell