adafruit/circuitpython #2876: Feather nRF52840 Sense and 2.4" TFT Featherwing - freezing display

MBottinIUT, 2020-05-10:
Hi,
I'm a little desperate and I'm hoping someone can give me an answer.
I'm trying to display bitmap images on the LCD screen of the '2.4" TFT Featherwing'. I use the displayio module, the ILI9341 driver and the latest version of Circuitpython (5.3.0).
I can display text and draw shapes, but I can't display images with 'OnDiskBitmap()'.

Everything works fine with Pyportal or with a Feather M4 Express + TFT Featherwing, but as soon as I run the same code on the Feather nRF52840 + TFT Featherwing, it doesn't work. The display freezes after a few lines of pixels.

I should point out that I use the same versions of Cicuitpython, libraries, code and image files in all my tests.

Please help me, because this is really out of my field of expertise !

Mike

Here is my code :
`import board
import time
import displayio
import terminalio
from adafruit_display_text import label
import adafruit_ili9341

# Release any resources currently in use for the displays
displayio.release_displays()

spi = board.SPI()
tft_cs = board.D9
tft_dc = board.D10

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=board.D6)
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

time.sleep(1)  # Wait a bit

splash = displayio.Group(max_size=10)
display.show(splash)

# Create a display group for our screen objects
g = displayio.Group()

# Display a ruler graphic from the root directory of the CIRCUITPY drive
f = open("/001.bmp", "rb")

pic = displayio.OnDiskBitmap(f)
# Create a Tilegrid with the bitmap and put in the displayio group
t = displayio.TileGrid(pic, pixel_shader=displayio.ColorConverter())
g.append(t)

# Place the display group on the screen
display.show(g)

# Refresh the display to have it actually show the image
display.refresh()
print("refreshed")
f.close()

while True:
    pass

`

deshipu, 2020-05-11:
One think that comes to mind that you could try is to set the speed of SPI explicitly after creating the SPI object:

```python
spi = board.SPI()
spi.configure(baudrate=20_000_000, phase=0, polarity=0)
```

MBottinIUT, 2020-05-11:
thank you very much for your proposal.
Unfortunately, when I test it, it doesn't work.
I've locked the SPI bus before add the configure method, but it doesn't work.
The code freezes and the REPL in Mu editor stays blank. I can't make any CTRL+C or CTRL+D.
I need to unplug and plug the board to get the hand back.

Do you have another idea ?

tannewt, 2020-05-11:
FourWire now takes `baudrate` in it's constructor so try it instead of changing the SPI setting.