adafruit/circuitpython #2880: gamepadshift: doesn't reliably respond to button presses anymore

jepler, 2020-05-11:
I believe this problem may have been introduced by the lower-power branch.

I don't have a simple how-to-reproduce, because the simple program I created to look for the problem seems to work just fine.  It may have to do with high CPU utilization from MP3 decoding, interrupts from sound DMA, etc., it's hard to be sure.

How to reproduce:
 * set up jeplayer and some mp3sa on a pygamer
 * with 5.3.0 loaded, start playing tracks, check that play/pause with the "any" button works
 * with master loaded, start playing tracks (you can usually get this far).  While playing, check that play/pause with "any" button works

For me, responding to the button click to play/pause is effectively instant in 5.3.0 and almost never happens with master.  Within the lower-power branch there are a lot of non-testable commits.  Here's my bisect log:
```
# bad: [6a5ab57c18b5a4ba1c80a951ec0ee655677f7d83] Merge pull request #2867 from jepler/stm32f4xx-rgbmatrix
git bisect bad 6a5ab57c18b5a4ba1c80a951ec0ee655677f7d83
# good: [51dbe9109fb6a474fe7f4cf4e3766e27ba0b34b5] Merge pull request #2797 from jepler/fix-2792
git bisect good 51dbe9109fb6a474fe7f4cf4e3766e27ba0b34b5
# bad: [755d404edf00ab99bb8497d2e4a36ba18dd17a69] Merge remote-tracking branch 'adafruit/master' into lower_power
git bisect bad 755d404edf00ab99bb8497d2e4a36ba18dd17a69
# bad: [17d038830d93946a2ed393376d62e33932153214] Use our own flag for board crystal config.
git bisect bad 17d038830d93946a2ed393376d62e33932153214
# good: [faa50df05fa11c3903934419a2ad7d7063cf23c7] Merge pull request #2817 from jepler/stm-linker-scripts
git bisect good faa50df05fa11c3903934419a2ad7d7063cf23c7
# skip: [a4d86b96fd1e60b807258a30cd765b48f0372b6d] Merge pull request #2777 from pewpew-game/master
git bisect skip a4d86b96fd1e60b807258a30cd765b48f0372b6d
# bad: [4b063aeb29c889809af0758351e68eda48e37f3a] Hack up STM32 PulseIn so it builds
git bisect bad 4b063aeb29c889809af0758351e68eda48e37f3a
# skip: [d9e68156b23cfca47592c5672e69622d6071ff60] fix up SAMD21 sleep
git bisect skip d9e68156b23cfca47592c5672e69622d6071ff60
# skip: [e4c4a4a3defa3ebb48bf4e139553243afdd84e2b] Swap to committed dependencies
git bisect skip e4c4a4a3defa3ebb48bf4e139553243afdd84e2b
# skip: [76ca13b6ed9d5729ee8c0a0d30d93c2af6cf7508] Fix SAMD51 pulsein
git bisect skip 76ca13b6ed9d5729ee8c0a0d30d93c2af6cf7508
# skip: [ea0ce39962d2dc8a1c188119a416b4916b9bd493] Simplify STM a bit.
git bisect skip ea0ce39962d2dc8a1c188119a416b4916b9bd493
# skip: [2623022c846698d8a6b9cfab8617aa569c5aef2f] Initial work for STM32. Need to fix us delay and PulseIn still.
git bisect skip 2623022c846698d8a6b9cfab8617aa569c5aef2f
# skip: [00d5f63e7cd3e489d3291ee7e90ab4ad46110b62] Hopefully fix PulseIn
git bisect skip 00d5f63e7cd3e489d3291ee7e90ab4ad46110b62
# skip: [6db11cf68ba2d5e17c6267f36f516bc0abce4643] Fix up Spresense build. It doesn't sleep.
git bisect skip 6db11cf68ba2d5e17c6267f36f516bc0abce4643
# skip: [affd3fcc2a0c577c58973c22af16383ff807a516] Clear the pending IRQ in the NVIC as well.
git bisect skip affd3fcc2a0c577c58973c22af16383ff807a516
# skip: [7100d5e485a93552e035476cb794145d6e194052] Fix autoreload and ticks in general
git bisect skip 7100d5e485a93552e035476cb794145d6e194052
# skip: [5e1e4d32db0e81bc414414ab60cef952db61b6c1] Get basic sleep going on the iMX RT 1011
git bisect skip 5e1e4d32db0e81bc414414ab60cef952db61b6c1
# skip: [4297ae2d0be7fd3b6c61a611c5598cb438b54b8f] Fix DotStar status LED init.
git bisect skip 4297ae2d0be7fd3b6c61a611c5598cb438b54b8f
# skip: [a8ed6d993cc6d1389017a9a1d38a1e830499b7c9] Switch iMX RT to RTC. Untested.
git bisect skip a8ed6d993cc6d1389017a9a1d38a1e830499b7c9
# skip: [c0c4c4403a24bc7c2bf2f12744cdc689be253842] Factor out stm32f4xx_hal_conf.h
git bisect skip c0c4c4403a24bc7c2bf2f12744cdc689be253842
# skip: [ed5cdd7e0970b904ed001deabe051c48ac32b1d3] Hopefully fix flash flush and hopefully audio as well.
git bisect skip ed5cdd7e0970b904ed001deabe051c48ac32b1d3
# skip: [a56170d7f8f4b807dcc72a1d3cce365d3523b124] Temporarily swap samd-peripherals location
git bisect skip a56170d7f8f4b807dcc72a1d3cce365d3523b124
# skip: [48b5f2a384e7ecc10ee51b453934a009ea9dc48d] Initial work on SAMD
git bisect skip 48b5f2a384e7ecc10ee51b453934a009ea9dc48d
# skip: [df5be65423e1cfd36af782108277c31cdb9056aa] Update FrequencyIn for new clock API.
git bisect skip df5be65423e1cfd36af782108277c31cdb9056aa
# skip: [798118b74f8ba408a6bbfe92d6ba925c68e388bf] Update board.c's to remove remaining tick.h
git bisect skip 798118b74f8ba408a6bbfe92d6ba925c68e388bf
# bad: [798118b74f8ba408a6bbfe92d6ba925c68e388bf] Update board.c's to remove remaining tick.h
git bisect bad 798118b74f8ba408a6bbfe92d6ba925c68e388bf
# good: [3a5f79acef1fd109aeb93e06e8a22dac5d54aa24] Merge pull request #2701 from hierophect/stm32-submodule-swap
git bisect good 3a5f79acef1fd109aeb93e06e8a22dac5d54aa24
# skip: [4fba4eb1331d28892a633517bc729811d0807e2a] Switch from 1k RTC input to 32k output
git bisect skip 4fba4eb1331d28892a633517bc729811d0807e2a
# skip: [653b7619ec60ff8167a52680d888f73062a8b06f] Add back microsecond delay to STM32
git bisect skip 653b7619ec60ff8167a52680d888f73062a8b06f
# skip: [418333979ae007b28855079c64ee3c65f76ecbc7] Fix autoreload, neopixel, monotonic_ns and sleep w/o SD
git bisect skip 418333979ae007b28855079c64ee3c65f76ecbc7
# skip: [6f60afe8c521011d1a2d34a2632295b483a8cee3] First try at lowering the power consumption
git bisect skip 6f60afe8c521011d1a2d34a2632295b483a8cee3
# only skipped commits left to test
```

caternuson, 2020-05-11:
Just to add - I can recreate this issue. Release 5.3.0 works fine. With version shown below from S3 bucket, behaves as described above, i.e. buttons essentially stop working.
```python
Adafruit CircuitPython 5.3.0-184-g90bd93180 on 2020-05-09; Adafruit PyGamer with samd51j19
>>> 
```

tannewt, 2020-05-11:
My guess is that this requires "supervisor_tick" to be called at a regular interval. I added internal enable/disable tick calls for this but didn't connect them up to gamepad or gamepadshift. So, these two classes should "enable_tick" until they are reset or deinit and then "disable_tick".

PR #2889 (2020-05-13): Gamepad, GamepadShift: Fix after lower-power by enabling supervisor tick
@tannewt your advice was right on, thanks.

Closes: #2880
closed 2020-05-13 by tannewt