adafruit/circuitpython #2887: RFC: Apply REUSE recommendations for license tagging

Flameeyes, 2020-05-12:
The [REUSE specifications](https://reuse.software) are a suggestion coming from FSFE to make code licensing information explicit and easier to apply. They build upon [SPDX Tags](https://spdx.org/) to make it obvious (and at the same time less verbose) which license apply to a certain work: code, documentation, examples, etc.

I have been [suggesting its use](https://flameeyes.blog/2020/04/23/reuse-simplifying-code-licensing/) and using it for my own projects, and I feel it would be a positive change for CircuitPython's examples in particular, as right now the easiest way to find which license they are released upon is to trust the repository-wide LICENSE file (rather than just looking at the snippets in the documentation.)

In addition to specifying _how_ to tag the licensing, the [reuse tool](https://github.com/fsfe/reuse-tool) is designed to aid adding the tags, and to validate the repository compliance to those specifications. Its linting mode can be integrated with both [pre-commit](https://pre-commit.com/) and with other CI tooling.

Because I was a bit over eager to help with this one, I already have a pull request that can show how this would look like for the repositories: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/99

In that pull request there's a few rough edges:

* All the code (and documentation) that otherwise didn't have an explicit license headers have been tagged with @ladyada's copyright as per the LICENSE file.
* The favicon.ico file has been tagged with MIT license and the same copyright information. I do believe this is incorrect, and would like to know what the correct information should be.
 * All configuration files have been tagged with ladyada's copyright and Unlicense. The current REUSE recommendation is to use CC0-1.0, but that has… side effects. There's some discussion in
fsfe/reuse-docs#62 about the recommendation for likely-uncopyrightable files (such as configuration files).

tannewt, 2020-05-13:
I think it's a good idea.

Would you mind updating our test repo including the CI checks? https://github.com/adafruit/Adafruit_CircuitPython_TestRepo

The step after that would be to update the cookiecutter: https://github.com/adafruit/cookiecutter-adafruit-circuitpython/

Flameeyes, 2020-05-13:
Oh that cookiecutter looks _nice_!

Sure I'll prepare a pull request for testrepo — there's at least one question that stays open though: what's the license of the logo? (In the favicon.ico)

tannewt, 2020-05-14:
The logo is CC-BY 4.0: https://www.dropbox.com/sh/l6tp9ym5nf8h5v9/AABGu_q2pcdUQdx1avr8Xvvda?dl=0&preview=license-readme.txt

It's worth noting that the copyright license on the logo doesn't override the limitations of using it because of Adafruit's trademark.

dhalbert, 2020-05-15:
@ladyada Do you or PT or your lawyers have an opinion on this? Also see
https://github.com/adafruit/Adafruit_CircuitPython_TestRepo/pull/33
https://github.com/adafruit/actions-ci-circuitpython-libs/pull/8

ladyada, 2020-05-18:
this is fine, please merge when ready and we'll keep an eye on it

Flameeyes, 2020-05-20:
To sum up current state:

 - https://github.com/adafruit/actions-ci-circuitpython-libs/pull/8 is needed to make `pre-commit run --all-files` pass when running as part of the CI;
 - once that's merged, https://github.com/adafruit/Adafruit_CircuitPython_TestRepo/pull/33 should pass CI and be ready to merge.

Should I wait for both to be merged before preparing the equivalent change for https://github.com/adafruit/cookiecutter-adafruit-circuitpython/, so we know what the final form of it should be?

tannewt, 2020-05-20:
@Flameeyes you are welcome to make the PR. we won't want to merge until it'll be green.

dhalbert, 2020-05-20:
> * [adafruit/actions-ci-circuitpython-libs#8](https://github.com/adafruit/actions-ci-circuitpython-libs/pull/8) is needed to make `pre-commit run --all-files` pass when running as part of the CI;
> * once that's merged, [adafruit/Adafruit_CircuitPython_TestRepo#33](https://github.com/adafruit/Adafruit_CircuitPython_TestRepo/pull/33) should pass CI and be ready to merge.

Both of these are now approved and merged.

Flameeyes, 2020-05-21:
Here's the cookiecutter PR: https://github.com/adafruit/cookiecutter-adafruit-circuitpython/pull/75

There's probably more opportunities for templating the copyright for the generated repository, do let me know if you notice any.

Flameeyes, 2020-05-30:
Current status:

 - cookiecutter PR needs review
 - https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/99/commits (the original RFC pull request) is updated to match the changes for TestRepo, including the updated code of conduct file, and the correct licensing for the favicon.

tannewt, 2020-06-01:
@Flameeyes Thanks for this work! I've merged both of these PRs and replied on the CircuitPython pre-commit PR.

dhalbert, 2023-10-28:
Note that in order to reduce skew from MicroPython, which does not use SPDX tags, we have removed SPDX from core files we inherit from MicroPython.