adafruit/circuitpython #2891: Xenon CircuitPython Crashing

desilveyk, 2020-05-13:
I have a Particle.io Xenon processor that I've converted to Circuit Python. I've installed the latest .udf2 file 5.3. After about 5 - 10 minutes or so the system crashes with the following error (see below). I'm running the adafruit gps_echotest.py example code. The putty session is closed once the error occurs. Any suggestions?

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.

jerryneedell, 2020-05-15:
How is your GPS connected? Uart or I2C? I can try to reproduce this.

desilveyk, 2020-05-15:
UART.

jerryneedell, 2020-05-15:
> UART. Sent from Outlook Mobile<https://aka.ms/blhgte>
> […](#)

OK - I have one running now - I have to go offline for a few hours but will report when I get back

jerryneedell, 2020-05-15:
I ran int for well over an hour with no crashes ...  Other tithe some periods of poor gps reception, it has been working fine. Just restarted about 5 min ago -- still no problems.

How are you powering the Xenon + GPS? I am just using USB from my host computer and have 3.3V from the ARGON to the GPS.

jerryneedell, 2020-05-15:
Just ran it for another hour with no problems.
Sorry, I am not able to reproduce the issue.

desilveyk, 2020-05-15:
I'm also running it from USB. Are you running it on a Aragon or Xenon processor? I'm running on a Xenon.

jerryneedell, 2020-05-15:
Yes, Xenon with an Ultimate GPS breakout board.

jerryneedell, 2020-05-15:
[image]

desilveyk, 2020-05-15:
Hum, maybe I just have a hardware issue on my processor. I have lots of them. I'll try the same conversation process on another one. Thanks very much for your time.

jerryneedell, 2020-05-15:
Glad to help -- Good luck!

closed 2020-06-22 by tannewt