adafruit/circuitpython #2893: microcontroller.cpu.voltage hangs on latest 5.4.0 RC

bd34n, 2020-05-14:
microcontroller.cpu.voltage hangs on latest 5.4.0 RC (5.4.0-beta.0-76-g0f55f5865-dirty), and only a hard reset appears to bring it back. Both .temperature and .frequency work, though.

This appears new - it didn't happen on 5.3.

Here is how to reproduce:

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.4.0-beta.0-76-g0f55f5865-dirty on 2020-05-14; BDMICRO Vina M4 v004 with samd51n20
>>> 
>>> import os
>>> import microcontroller
>>> 
>>> os.uname()
(sysname='samd51', nodename='samd51', release='5.4.0', version='5.4.0-beta.0-76-g0f55f5865-dirty on 2020-05-14', machine='BDMICRO Vina M4 v004 with samd51n20')
>>> 
>>> microcontroller.cpu.temperature
39.0517
>>> 
>>> microcontroller.cpu.frequency
120000000
>>> 
>>> microcontroller.cpu.voltage
[locked up at this point]

I apologize that I don't have time to attach my j-link and trace it. But thought it worthwhile to at least file the issue.

siddacious, 2020-05-18:
Can't reproduce on an m0 so it may be an M4 specific thing?
[image]

dhalbert, 2020-05-18:
The SAMD51 ADC is odd and flaky in a number of ways, sadly, so not so surprising.

DavePutz, 2020-06-14:
Disclaimer: All I know about SAMD51 ADC's is what I picked up while looking at this issue.
1) This is not just 5.4 related. I was able to reproduce on a PyBadge back to CP 5.0
2) The hang is in ports/atmel-samd/asf4/samd51/hal/src/hal_adc_sync.c, line 121:
'''
while (!_adc_sync_is_channel_conversion_done(&descr->device, channel))
			;
'''
This can loop forever. I added code to put in a loop counter and exit after some number of attempts. While this eliminated the hangs it did not resolve the real issue. It does, perhaps,
look like timing related; since when I added a print statement in the loop the first read being done (from common_hal_mcu_processor_get_voltage()  line 312 ) exited after hitting the loop limit, but the second read then succeeded on the first attempt. My current code, which works but is an enormous kludge, is as follows:
'''
uint8_t  loop_ctr = 0;
                while (!_adc_sync_is_channel_conversion_done(&descr->device, channel)) {
                        loop_ctr++;
                        mp_printf(&mp_plat_print, "");
                        if (loop_ctr > 50) {
                           return 0;
                        }
                 }
'''
So, it would appear that the print statement does enough to let the adc_sync_is_channel_conversion_done() test succeed. There must be a better way - any suggestions?

DavePutz, 2020-06-15:
So, testing shows that if the loop limit is increased to 255 and the mp_printf is removed then Python calls to microprocessor.get_voltage() will succeed. @dhalbert , that would require making changes to the asf4 library. Is that something we could/would do?

dhalbert, 2020-06-15:
We have already forked asf4 and fixed a small number of things, so there's no problem. I haven't had a chance to look at this in detail yet.

dhalbert, 2020-06-15:
We may be forgetting some sync-wait we need during the setup of the ADC. We may not need to change the ASF4 code.

dhalbert, 2020-06-19:
I have a number of SAMD51 boards, including two Metro M4 boards that are identical. There has been a chip revision, but the SAMD51 chips on them have identical revision numbers in the DSU DID device field. They vary only in the date code on the package.

On one, `microcontroller.cpu.voltage` hangs, and on the other, it does not. So this varies per chip sample (ugh), and I'll investigate further.

I also tested several other SAMD51 boards and `.voltage` doesn't hang. So it's less-common problem.

PR #3052 (2020-06-22): delay 1ms on SAMD51 when reading processor voltage
Fixes #2893 

On some SAMD51 processor samples that are seemingly identical to others, reading the CPU voltage can hang the ADC. This appears to be due to `VREF.VREFOE` startup time. Adding a small delay fixes the problem. I did not attempt to tune the delay to a minimum value, since it might vary based on sample.

I was able to remove the old code setting `VREF.ONDEMAND` after adding the delay.

See https://community.atmel.com/forum/samd51-using-intref-adc-voltage-reference for an Atmel support community thread I started about this.

Tagging  [... cut]
closed 2020-06-22 by tannewt