adafruit/circuitpython #2894: Pulsewidth "deinit" does not complete

RetireeJay46, 2020-05-14:
I'm using a Metro Express M0 board with CircuitPython 5.2.0. I have built my own LED driver "shield" that simply buffers digital outputs from the M0. I am trying to write a CircuitPython program that will create a "walking" sequence of up to 5 LED's, i.e. one LED turns on then off; when the first one turns off the second one turns on, and so on. I also want to be able to vary the frequency. So in order to guarantee correct timing from LED 1 to LED 2 to LED 3, etc. my concept is to completely de-initialize all Pulsewidth outputs before changing the frequency, and then setting up the Pulsewidth outputs all over again at the new frequency.
The demo program here, when run, shows what happens (with only 3 LED's). It successfully starts up (Trial 1), and it successfully changes frequency one time, but when it starts to run the next round (Trial 3) it crashes saying that "All timers are in use". I expected that the carefully programmed "deinit" function calls should have released all timers for re-use, but that's not happening - even though the program reports successful deinit every time.

[DemoDeinitNotWorking.txt](https://github.com/adafruit/circuitpython/files/4631391/DemoDeinitNotWorking.txt)

DavePutz, 2020-05-18:
I may be mis-interpreting here; but it is my understanding that with LEDs on PWM you want to set the frequency once (basically high enough to avoid flicker) and then modify the duty cycle to change the brightness. I don't think you need to keep creating and de-initing the PWMOut objects - just create them once and then keep altering the pwn.duty_cycle to vary the brightness.

siddacious, 2020-05-18:
@DavePutz I believe they are using PWM at a (relatively) low frequency to do the sequencing of the LEDs instead of doing the timing themselves with `time.sleep` or `time.monotonic`. I don't don't think the intent is to adjust the brightness. 

@RetireeJay46 Am I correct? It's not a _typical_ use of pwm with leds, but I don't see why it shouldn't work. Certainly `deinit()` should be releasing the timers so that seems like a bug.

Here's my run, modified to run on a CPX with the latest build of `master`:
[image]

DavePutz, 2020-07-13:
Testing on CP 6.0.0-alpha.1; the issue no longer reproduces for me.

tannewt, 2020-07-13:
@RetireeJay46 Can you try 6.0.0-alpha.1 and let us know if it works now? Thanks!

RetireeJay46, 2020-07-14:
It now works on my system.  Thanks!!

closed 2020-07-14 by RetireeJay46