adafruit/circuitpython #2912: time.sleep() in code.py prevents USB from working

dhalbert, 2020-05-16:
This may be a dupe of the other USB issues we are seeing with `lower_power`.

If this is `code.py`, USB doesn't connect: no CIRCUITPY and no REPL
```python
import time
while True:
    print('hi')   # not necessary, just easier to see it's running
    time.sleep(1)
```

xobs, 2020-05-18:
This is a duplicate of https://github.com/adafruit/circuitpython/issues/2855 and is fixed by https://github.com/adafruit/circuitpython/pull/2868

dhalbert, 2020-05-18:
I neglected to say I was seeing this issue on SAMD21 boards (CPX and Metro M0). I could not reproduce it on SAMD51, though it's probably an accident that it works :). The #2868 fix is specific to `nrf`, so we'll need fixes for other ports too.

PR #2956 (2020-05-26): fix racing issue for SAMD when executing WFI
fix #2912 and/or other "slow responses" with USB after low power PR. It is basically the same as #2868 for nRF. There is lots of discussion there, I will sum it up here. Following is the racing.

1. USB Interrupt handler  is complete and submitted to the event queue of TinyUSB task
2. WFI happens
3. Since `tud_task()` isn't called yet to response, there is no more USB event. CPU has a good full sleep of `N` seconds.
4. WFI end, tud_task() is called to handle usb events

It is the racing between usb isr -> event queue -> tud_task() handling and WFI.
 [... cut]
closed 2020-05-26 by tannewt