adafruit/circuitpython #2916: No Module Named analogio

ArmstrongSubero, 2020-05-16:
When using the STM32F746ZG discovery get "ImportError: no module named 'analogio'"

ArmstrongSubero, 2020-05-18:
Digging through the source saw it is not supported. Maybe when I get some time if no one gets around to it I'll do a port.

closed 2020-05-18 by ArmstrongSubero

tannewt, 2020-05-18:
@ArmstrongSubero Let us know if you need help when adding it. We love getting more contributors involved.

ArmstrongSubero, 2021-05-17:
I added it. 

https://github.com/ArmstrongSubero/CircuitPython/tree/main/STM32F746ZG%20Analog%20Support/build-nucleo_f746zg

ladyada, 2021-05-17:
@ArmstrongSubero nice work! next up is a PR, do you need a hand creatinng one?

ArmstrongSubero, 2021-05-17:
@ladyada Thank you! I'll do a PR in about a week, I want to do some more testing first 😁

hierophect, 2021-05-17:
@ArmstrongSubero a draft PR is a good place to start - it lets other people check things out but keeps it from being merged if there are other things you want to make sure are added. I have a STM32F746ZG and worked on the F4 AnalogIO previously so if you decide to post a PR I'd be happy to give it a test.

ArmstrongSubero, 2021-05-17:
@hierophect np I'm swamped today so I'll do one tomorrow.

ArmstrongSubero, 2021-05-19:
@hierophect I did the PR.