adafruit/circuitpython #2923: Add internal links to boards in the support matrix

tannewt, 2020-05-18:
So we can link folks to a particular line of the matrix.

https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html

askpatrickw, 2020-07-22:
I thought I'd help out and knock this one out which was a fail. Here's some notes for the next person:

`docs/shared_bindings_matrix.py` is used to run the Jinja2 template syntax within `shared-bindings/support_matrix.rst`.

Within `shared-bindings/support_matrix.rst` the Jinja for loop is used to basically pass a CSV formatted list into the csv-table RestructuredText directive.

```python
{% for key, value in support_matrix|dictsort -%}
"{{ key }}", "{{ '`' ~ value|join("`, `") ~ '`' }}"
{% endfor -%}
```

Where I failed is that I could not come up with a way to insert either the standard RST `:ref:` or HTML syntax to create an ID on the Cell or a div around the key into the Jinja for loop which would be interpreted properly. Everything was displayed as the cell content.

I believe this is just the limitation of the csv-table directive in RestructuredText. But I'm not 100% positive about that. An RST guru's input would be helpful.

sommersoft, 2020-08-16:
Fixed by #3256

closed 2020-08-16 by sommersoft