adafruit/circuitpython #2944: esp32s2 CIRCUITPY drive not accessible from Linux

jerryneedell, 2020-05-22:
Feel free to close if it is too early for issues on this - I just wanted to give you a heads up.

Using Ubuntu 20.04 -- when I plug in the esp32s2 its boots and CIRCUITPY mounts -- REPL works

But if I try to access the CIRCUTPY drive -- Linux cannot read it.

```
it gets very upset ``` jerryneedell@jerryn#2399eedell-ubuntu-macmini:~$ ls /media/jerryneedell/CIRCUITPY/
ls: cannot access '/media/jerryneedell/CIRCUITPY/    FSEV.EN~': Input/output error
ls: cannot access '/media/jerryneedell/CIRCUITPY/    '$'\001''.': No such file or directory
ls: cannot access '/media/jerryneedell/CIRCUITPY/e.DA~': Input/output error
ls: cannot access '/media/jerryneedell/CIRCUITPY/    TRAS.HE~': Input/output error
ls: cannot access '/media/jerryneedell/CIRCUITPY/äI'$'\002''.Be': Input/output error
ls: cannot access '/media/jerryneedell/CIRCUITPY/äI.A.': No such file or directory
ls: cannot access '/media/jerryneedell/CIRCUITPY/  .TRA': Input/output error
ls: cannot access '/media/jerryneedell/CIRCUITPY/äi'$'\003''.boo': Input/output error
''$'\001'       '    '$'\001''.'   äI.äI              boot_out.txt    'EN~1   '$'\020'   n.MET
''$'\b'         'ä-²?.'$'\001'     äI.äI             ''$'\001''.cir'       FSEV.EN~        .TRA
''$'\030'        a.e              'äI'$'\002''.Be'    DA~1             HE~1                  TRAS.HE~
'   '$'\020'    'äI'$'\004'       'äi'$'\003''.boo'   e.               lib              'TXT '$'\030''.äI'
'   '$'\020\b'   äI.A.             äI.LIB             e.              'l-ⁿ?.'$'\001'    'UITPY  '$'\b'
'['             'äi.äi'$'\004'     a.n                e.DA~            n                'Y  '$'\b'
jerryneedell@jerryn#2399eedell-ubuntu-macmini:~$ 
```

dmesg shows
```
[273479.668050] sd 7:0:0:0: [sdc] 4097 512-byte logical blocks: (2.10 MB/2.00 MiB)
[273479.697795] sd 7:0:0:0: [sdc] Write Protect is off
[273479.697802] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[273479.727778] sd 7:0:0:0: [sdc] No Caching mode page found
[273479.727789] sd 7:0:0:0: [sdc] Assuming drive cache: write through
[273479.947858]  sdc: sdc1
[273480.117956] sd 7:0:0:0: [sdc] Attached SCSI removable disk
[273483.087851] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[273483.367782] FAT-fs (sdc1): error, corrupted directory (invalid entries)
[273483.367788] FAT-fs (sdc1): Filesystem has been set read-only
[273483.368366] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[273486.459292] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[273486.460088] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[273486.460179] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[273486.460318] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 0)
[273486.460542] FAT-fs (sdc1): error, corrupted directory (invalid entries)
[273486.460677] FAT-fs (sdc1): error, corrupted directory (invalid entries)
jerryneedell@jerryneedell-ubuntu-macmini:~$ ls /media/jerryneedell/CIRCUITPY/
```
REPL see the file system OK
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.4.0-beta.0-291-geea01c32b-dirty on 2020-05-22; Saola 1 w/Wrover with ESP32S2
>>> import os
>>> os.listdir()
['.fseventsd', '.metadata_never_index', '.Trashes', 'lib', 'boot_out.txt']
>>> 
```

BTW
I plugged it into my Mac and it works fine!  CIRCUITPY mounts and I can read it.

jerryneedell, 2020-05-23:
Very interesting --

If I flash the board with CP on the linux box - I am not able to access the CIRCUITPY drive as above BUT if I then plug it into my Mac I can access it fine and then if I go back to the Linux box it also works normally - I can read/write to the CIRCUITPY drive.

summary
flash CP to board  on linux -- CIRCUITPY not readable
plug into Mac -- CIRCUITPY OK
plug back into Line -- CIRCUITPY OK

geekguy-wy, 2020-05-23:
I have not seen this on Raspbian Buster or Ubuntu 18.04.

jerryneedell, 2020-05-23:
@geekguy-wy with an esp32-s2?

geekguy-wy, 2020-05-23:
Oh, no, not with that one, which I do not have, sorry.

jerryneedell, 2020-05-23:
hmm -- a bit more information -  Once the File system has been "fixed" by plugging it into the Mac, I am able to refresh CP to the board and the FS (CIRCUITPY) is still accessible on the Linux (and a Raspberry Pi) however, if i erase the flash then refash CP, it is back to the bad state that cannot be accessed on the Linux box (or raspberry Pi) but can be "fixed" but plugging it into the Mac.

I hope that makes sense...

I think this makes some sense since if I recall correctly flashing CP does not destroy an existing File system. Erasing the board does.  It looks like there is something about plugging git into the Mac that "corrects" some issue with the FS.

jerryneedell, 2020-05-23:
> Oh, no, not with that one, which I do not have, sorry.

no problem -- that makes me feel better -- I think ;-)

jerryneedell, 2020-05-28:
FYI -  On recent attempts, when re-flashing CP to the board, the FS becomes inaccessible from Linux. ( as noted above -- it seemed like I was able to access it on some previous occasion)
When I plug it into the Mac it is recovered and the files on the CIRCUITPY drive are once again available to Linux, that is, the FS is not corrupted or erased, but I do have to go through the step of plugging it into the Mac before I can use it.

latest try with ```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.4.0-beta.0-380-gf72d166e5 on 2020-05-28; Saola 1 w/Wrover with ESP32S2
>>>

sbehnke, 2020-05-28:
I can also confirm this is happening with my esp32 soala s2.

tannewt, 2020-05-28:
Ok, I think I have a fix for this in my `esp32s2_busio` branch. The memcpy was copying too much into the cache (therefore overwriting with garbage.) I'll split it out today or tomorrow with a couple other fixes and get it PRed.

closed 2020-05-28 by dhalbert

sixtyfive, 2021-04-22:
Now the same thing is happening to me with a QtPy and CircuitPython 6.2.0. Very strange.

PR #2972 (2020-05-28): A number of small ESP32S2 fixes
* Update ESP IDF and change USB init accordingly.
* Fix flash writes that don't end on a sector boundary. Fixes #2944
* Fix enum incompatibility with IDF.
* Fix printf output so it goes out debug UART.
* Increase stack size to 8k.
* Fix sleep of less than a tick so it doesn't crash.
commit 796373b8 (2020-05-28): A number of small ESP32S2 fixes:
* Fix flash writes that don't end on a sector boundary. Fixes #2944
* Fix enum incompatibility with IDF.
* Fix printf output so it goes out debug UART.
* Increase stack size to 8k.
* Fix sleep of less than a tick so it doesn't crash.