adafruit/circuitpython #2946: NRF: nRF52833 needlessly allocates 8 kB of RAM to SPIM3

xobs, 2020-05-23:
As noted by @dhalbert in https://github.com/adafruit/circuitpython/pull/2933#issuecomment-632859678, the new nRF52833 part does not suffer from the same DMA bug that requires the nRF52840 port to dedicate 8 kB of RAM to SPIM3.

Currently, the one nRF52833 port (Simmel) disables SPIM3, which means it does not have access to the sole fast SPI bus in the system.

The SPIM3 DMA workaround should be special-cased so that it does not get enabled on parts that are not nRF52840.

xobs, 2020-05-23:
I'm mostly creating this issue so that I don't forget about it!

dhalbert, 2020-07-13:
> Relatedly, the current linker file only correctly implements the required workaround if `SOFTDEVICE_RAM_SIZE` is a multiple of 8K of at most 56K. 

Right, this should be noted in a comment. The `SOFTDEVICE_RAM_SIZE` was determined by trial and error, and I always meant it to be a multiple of 8K.

> Also, I believe
> 
> https://github.com/adafruit/circuitpython/blob/37e77b21cd3e33b00675310e4070e115d3c2357c/ports/nrf/boards/common.template.ld#L44
> 
> should be `+ LENGTH(SPIM3_RAM)`.

That is correct. Fortunately, `_spim3_ram_end` isn't actually used anywhere. :smiley:

riggs, 2020-07-13:
Do note that 64K of softdevice RAM (current default) puts the SPIM3 area in the same block as all the other application RAM.

dhalbert, 2020-07-13:
@riggs You are right, I see what you mean. I missed the non-uniform size of the RAM blocks. I will open a new issue for this.

PR #3281 (2020-08-17): SPIM3 buffer must be in first 64kB of RAM
Fixes #3247 and fixes #3250 (@DavePutz, please test, thanks)
Fixes #3147 
Fixes #2946 

#3250: I believe this error is caused by #3147, which notes that the SPIM3 buffer is misplaced. To put it in its own 8kB region, the SoftDevice RAM area had to shrink from 64kB to 56kB. This was done by shrinking `BLEIO_HVN_TX_QUEUE_SIZE` from 9 to 5. The number of connections did not need to be reduced from its present 5.

For nRF52833, the special SPIM3 buffer is not necessary, because the SPIM3 peripheral does not suffer from the same hardware bug as on the nRF52 [... cut]
closed 2020-08-17 by dhalbert