adafruit/circuitpython #2950: REPL doesn’t have exit or quit functions

merwok, 2020-05-23:
Hello! Many people call `exit()` or `quit()` at the regular Python prompt rather than Ctrl-D or other key combos on other operating systems.  The CircuitPython REPL (as seen from the Mu editor) doesn’t have exit or quit.  (I also tried `import site` to have them added but could not import.)

IMO it is easier to document calling a function than a shortcut that may be unfamiliar, changes between platforms (I think macs don’t have Ctrl), can be confused with Ctrl-C or Ctrl-Z.

(Redirected here from https://github.com/mu-editor/mu/issues/1046)

merwok, 2020-05-23:
For reference, I have now found a rejected feature request from upstream: https://github.com/micropython/micropython/issues/893#issuecomment-58062897
But maybe Adafruit wants to make a different decision for their version :slightly_smiling_face:

deshipu, 2020-05-24:
Where would it exit to? There is no underlying operating system, CircuitPython is the only thing running.

kmatch98, 2020-05-24:
Maybe supervisor.reload() is what you are looking for, similar to CTRL-D?

https://circuitpython.readthedocs.io/en/3.x/shared-bindings/supervisor/__init__.html

merwok, 2020-05-25:
It would exit in the same way that Ctrl-D exits, i.e. terminate the process.
This is desirable behaviour for Mu editor: the same pane switches from attached device stdout and REPL.  People can run a script, see its output, click to open a REPL, look at things, Ctrl-D to go back to stdout, edit code and save to see new output.

tannewt, 2020-05-26:
I'm open to supporting one but not both so that docs are clearer. Perhaps we should create a `site` module and only auto-import it into the REPL. @merwok are you interested in adding it? I'm happy to give guidance on how to do it.

tannewt, 2020-05-26:
Also, doc for these functions in CPython is here: https://docs.python.org/3/library/constants.html?highlight=exit#exit

merwok, 2020-05-26:
Why add one and not the other?  The Python stdlib has had both for ages, I suppose to be compatible with expectations from other systems, or people’s intuition.

I’m interested in making a PR.  Just tell me where `site.py` should be added in the source!

tannewt, 2020-05-26:
> Why add one and not the other? The Python stdlib has had both for ages, I suppose to be compatible with expectations from other systems, or people’s intuition.

Generally it's better only have one way to do something. In this case I guess both are ok since it's meant to make it easy to escape the REPL.

> I’m interested in making a PR. Just tell me where `site.py` should be added in the source!

Unfortunately, I don't think it's that easy. We'll need to add a `site` module to shared-bindings and the impl to shared-module. See `struct` for an example: https://github.com/adafruit/circuitpython/tree/master/shared-bindings/struct

Then, the trickiest bit is auto-importing it into the REPL's top level namespace. I think it'd be done here: https://github.com/adafruit/circuitpython/blob/master/main.c#L402 But not sure what it'd exactly take.

dhalbert, 2020-05-26:
A whole new module is going to use up more flash. I was thinking that just adding some global functions which point to the same thing might take less.

dunkmann00, 2020-05-27:
https://github.com/adafruit/circuitpython/blob/739b42e3e0cde489a5deadb945ad4e754fa75baf/lib/utils/pyexec.c#L470-L474

What if you changed the conditional that checks for Ctrl-D to also check for if the user types `quit()`? 

i.e.:
`} else if (ret == CHAR_CTRL_D || (strcmp(vstr_str(&line),"quit()") == 0)) {`

tannewt, 2020-05-27:
@dunkmann00 That's a great idea! I'm not used to REPL land where we have each line. :-)

dhalbert, 2020-05-27:
`quit    (     )` won't work, etc. It does work in CPython.

dhalbert, 2020-05-27:
I didn't mean to be curt. In CPython, `sys.exit()` raises `SystemExit` and quits. It appears that `exit()` and `quit()` are just aliases for `sys.exit()` via the `site` module. You can change them if you want.

So I would suggest we either provide the `site` module, or we just make the builtin functions `exit()` and `quit()` be aliases for `sys.exit()`.

Note that `sys.exit()` is not working right in CircuitPython right now. It just goes on to the next REPL line.

merwok, 2021-03-06:
I think the `site` module adds these two pseudo-built-in functions so that people beginning with Python can type `exit` without parens get a useful message.  Without that, I’m not sure that these functions have enough value to be added.