adafruit/circuitpython #2952: Build from Source not Working

ArmstrongSubero, 2020-05-24:
Maybe it's my inexperience with this build system. Able to build for STM32F746GZ and SAMD21G18A however after building flashing HEX/BIN or UF2 in bootloader mode and the device does not show up as usb device. On both windows and unix the device will show up after about 5 mins, and in windows it fails with USB descriptor error. 

Is it a clock error or something? Cause if I load the build output on the board it does run after loading the UF2 file. The problem is that it does not show up as a USB device.

deshipu, 2020-05-25:
Is this for a custom board of yours, or for existing board definitions?

ArmstrongSubero, 2020-05-25:
Well the Nucleo 746ZG board is the same.

The SAMD21G18 is custom, but set up just as Arduino Zero (the UF2 bootloader was sent to it as Arduino Zero and shows up as 'ZEROBOOT'

deshipu, 2020-05-25:
Could you try compiling your SAMD21 firmware as crystalless as a test? That would let us rule out clock problems, I think.

ArmstrongSubero, 2020-05-25:
Thanks for your help! I got it to work by adding

#define CALIBRATE_CRYSTALLESS 1
or
#define BOARD_HAS_CRYSTAL 0

in my myconfigboad.h, I have a 32.768 crystal connected so both options work.

Can you tell me where I will be able to see a list of all options available, both for the myconfigboard.h and the mpconfigboard.mk?

ArmstrongSubero, 2020-05-25:
so I was right it was a clock error, if USB takes so long to show up.

One more thing, I see its missing the "code.py" file, is there a frozen module or something I must add to get the code.py file to be added automatically?

ArmstrongSubero, 2020-05-25:
unfortunately that solution does not work for the STM32F746ZG target, it doesn't show up as USB

tannewt, 2020-05-26:
@ArmstrongSubero I recommend joining the Adafruit Discord. It's easier to give support there. URL to join is https://adafru.it/discord

hierophect, 2020-05-28:
@tannewt @ArmstrongSubero missed this issue. Highly likely it's related to the low power domain which I managed to fix for the F4s but may have had some straggler issue for the F7 series. I'll check it out as soon as I can.

k0d, 2020-05-29:
On the Nucleo-F746G I have no issues with booting when I try the 5.4 beta or the most recent S3-build.

hierophect, 2020-06-01:
I'm also unable to replicate on 5.4. If you're using an old build, it might be related to #2866. In any case, I've modified the crystal settings in #2979 - if 5.4 still isn't working for you, you could give it a shot.

Closing for now since we aren't able to reproduce it, and it seems like the Atmel problem is possibly a custom board layout issue. Please do feel free to update though if you continue having any problems on the STM32 boards, we're trying to work out the kinks in the new low power system.

closed 2020-06-01 by hierophect

ArmstrongSubero, 2021-05-17:
I managed to get it to compile and CircuitPython works with analog support. I've been using 5.x build though. Anyone having similar problems can use:

https://github.com/ArmstrongSubero/CircuitPython/tree/main/STM32F746ZG%20Analog%20Support/build-nucleo_f746zg