adafruit/circuitpython #2958: ESP32S2 time.sleep() issues

brandonvfx, 2020-05-26:
While I was testing out the code for #2931 and found some issues with how `time.sleep()` is working for the ESP32S2.

1. Float values passed seem to be rounded up to the nearest integer.
2. Passing a small float value like `0.001` causes a board to crash on boot (serial output below). Since the board no longer mounts because of that, only solution I was able to find to get a board working again was to run `idf.py erase_flash` then reflash CP.

Hardware tested on:
- Espressif Saola Wrover
- A custom development board

Commit for the build I was using:
```
commit 3bb6a32a2078452d3278b185a15721391d78a9db (HEAD, tannewt/esp32s2_digitalio)
```

Serial output after hard reset (Saola):
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x1914
load:0x40050000,len:0x14b0
load:0x40054000,len:0x210c
entry 0x400502d8
I (48) boot: ESP-IDF v4.2-dev-1398-g0daf6e0e4 2nd stage bootloader
I (48) boot: compile time 13:18:19
I (48) boot: chip revision: 0
I (52) boot.esp32s2: SPI Speed      : 40MHz
I (56) boot.esp32s2: SPI Mode       : DIO
I (61) boot.esp32s2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (90) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (97) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (104) boot:  3 ota_0            OTA app          00 10 00010000 00080000
I (112) boot:  4 ota_1            OTA app          00 11 00090000 00080000
I (120) boot:  5 user_fs          Unknown data     01 81 00200000 00200000
I (127) boot: End of partition table
I (131) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x0d0d0 ( 53456) map
I (155) esp_image: segment 1: paddr=0x0001d0f8 vaddr=0x3ffc1130 size=0x01a24 (  6692) load
I (158) esp_image: segment 2: paddr=0x0001eb24 vaddr=0x40028000 size=0x00404 (  1028) load
I (162) esp_image: segment 3: paddr=0x0001ef30 vaddr=0x40028404 size=0x010e8 (  4328) load
I (172) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x2b094 (176276) map
I (229) esp_image: segment 5: paddr=0x0004b0bc vaddr=0x400294ec size=0x07c38 ( 31800) load
I (244) boot: Loaded app from partition at offset 0x10000
I (245) boot: Disabling RNG early entropy source...
I (246) cache: Instruction cache        : size 16KB, 4Ways, cache line size 32Byte
I (253) cpu_start: Pro cpu up.
I (256) cpu_start: Application information:
I (261) cpu_start: Project name:     circuitpython
I (267) cpu_start: App version:      5.3.0-140-g3300495bb
I (273) cpu_start: Compile time:     May  9 2020 13:15:03
I (279) cpu_start: ELF file SHA256:  6d6940b9a599e974...
I (285) cpu_start: ESP-IDF:          v4.2-dev-1398-g0daf6e0e4
I (291) cpu_start: Single core mode
I (296) heap_init: Initializing. RAM available for dynamic allocation:
I (303) heap_init: At 3FFD7400 len 00024C00 (147 KiB): D/IRAM
I (309) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (315) cpu_start: Pro cpu start user code
W (375) CircuitPython: Hello from CircuitPython
W (375) CircuitPython: port init done
W (1365) CircuitPython: port fixed stack
W (1365) CircuitPython: got limit 0x3ffd89c0
W (1365) CircuitPython: got length 3584
W (1375) CircuitPython: port fixed stack
W (1375) CircuitPython: got limit 0x3ffd89c0
W (1385) CircuitPython: got length 3584
W (1385) CircuitPython Internal Flash: fatfs partition 0x3fffc2a4
W (1395) CircuitPython Internal Flash: read 0
W (1395) CircuitPython Internal Flash: read 0
W (1405) CircuitPython Internal Flash: read 0
W (1405) CircuitPython Internal Flash: read 0
W (1415) CircuitPython: port fixed stack
W (1415) CircuitPython: got limit 0x3ffd89c0
W (1425) CircuitPython: got length 3584
W (1425) CircuitPython: port fixed stack
W (1435) CircuitPython: got limit 0x3ffd89c0
W (1435) CircuitPython: got length 3584
W (1445) CircuitPython Internal Flash: read 0
W (1445) CircuitPython Internal Flash: read 0
../../esp-idf/components/freertos/tasks.c:1339 (vTaskDelayUntil)- assert failed!

abort() was called at PC 0x4002cd7a on core 0

Backtrace:0x4002b4ff:0x3ffd94f0 0x4002bb35:0x3ffd9510 0x4002e94e:0x3ffd9530 0x4002cd7a:0x3ffd95a0 0x400982df:0x3ffd95c0 0x40098aa8:0x3ffd95e0 0x400a2429:0x3ffd9600 0x400a1955:0x3ffd9620 0x4008dda9:0x3ffd9640 0x4008aab9:0x3ffd9660 0x4008abb6:0x3ffd9680 0x400961be:0x3ffd96a0 0x4008de6d:0x3ffd9740 0x4008aab9:0x3ffd9790 0x4008aae6:0x3ffd97b0 0x4009f8c7:0x3ffd97d0 0x4009fb8a:0x3ffd9880 0x40097e05:0x3ffd98a0 0x40097e98:0x3ffd98d0 0x40098110:0x3ffd9920 0x400982f8:0x3ffd9940 0x400a4c40:0x3ffd9960 0x4002bb3d:0x3ffd9990

ELF file SHA256: 6d6940b9a599e974

CPU halted.
```

jerryneedell, 2020-05-26:
FYI -- I have noticed the issue 1) with time.sleep()
I was trying a simple "blinky" -- using time.sleep(.25) as the delay between blinks appears to blink at the same rates as if I use time.sleep(1.0).
using saola_wrover

jerryneedell, 2020-05-29:
Is this due to Floating point not being enabled?
https://github.com/adafruit/circuitpython/blob/master/shared-bindings/time/__init__.c#L66

tannewt, 2020-05-29:
My guess is that it's related to the conversion from CP ticks to FreeRTOS ticks here: https://github.com/adafruit/circuitpython/blob/master/ports/esp32s2/supervisor/port.c#L146

I believe the FreeRTOS ticks are at the 1/10th of a second resolution so we'll also need to account for the remaining time.

DavePutz, 2020-06-07:
The issue actually looks to be in https://github.com/adafruit/circuitpython/blob/master/ports/esp32s2/supervisor/port.c#L118 . The code ends up dividing tv_usec by 32768 to get all_subticks, and then divides that by 32 before adding it in the return statement. That is equivalent to dividing by 1048576. Since tv_usec seems to run from 1 to 1,000,000; the result is that nothing gets added to tv_sec and the return from port_get_raw_ricks(NULL) is always going to be just seconds.
So, in a Python script, sleep(1.1) will sleep for 2 seconds; while sleep(0.9) will sleep for 1 second.
I am not sure what needs to be changed, however. If port_get_raw_ticks() is modified to be:
``
    uint64_t all_subticks = (uint64_t)tv_now.tv_usec / 1000;
    if (subticks != NULL) {
        *subticks = all_subticks % 32;
    }
    return (uint64_t)tv_now.tv_sec * 1024L + all_subticks ;
``
then sleep() seems to work correctly; but there may be effects in other code that needs to be accounted for.

tannewt, 2020-06-08:
@DavePutz Would you mind opening a PR for this? I think you are on the right track. The raw tick count should be in 1/1024ths of a second. A subtick is 1/32nd of that. (Most boards use a 32.768khz clock for timing 1 second.)

tannewt, 2020-06-22:
@brandonvfx Can you test the latest version? It should be more accurate thanks to @DavePutz

brandonvfx, 2020-06-23:
I tested `main` on a Saola and my board, and it looks like both issues have been resolved. Thanks! 😃

closed 2020-06-23 by brandonvfx