adafruit/circuitpython #2962: Automate USB VID/PID uniqueness check

tannewt, 2020-05-27:
We should add a CI check for USB VID/PID validity. It's a common issue with new board PRs

sommersoft, 2020-05-27:
This one isn't too difficult, but its going to require either some additional metadata or cleaning up prior to implementation:
```
Duplicate found:
        Boards: stm32f411ce_blackpill, thunderpack
        VID:PID: 0x239A:0x806A
Duplicate found:
        Boards: meowbit_v121, feather_stm32f405_express
        VID:PID: 0x239A:0x805A
Duplicate found:
        Boards: makerdiary_nrf52840_mdk, feather_nrf52840_express, makerdiary_nrf52840_mdk_usb_dongle, 
                TG-Watch02A, pca10100, pca10059, pca10056
        VID:PID: 0x239A:0x802A
Duplicate found:
        Boards: electronut_labs_blip, electronut_labs_papyr
        VID:PID: 0x239A:0x803C
Duplicate found:
        Boards: pewpew10, gemma_m0
        VID:PID: 0x239A:0x801D
Duplicate found:
        Boards: feather_m0_supersized, feather_radiofruit_zigbee, feather_m0_express_crickit, feather_m0_express
        VID:PID: 0x239A:0x8023
Duplicate found:
        Boards: trinket_m0, ugame10, trinket_m0_haxpress
        VID:PID: 0x239A:0x801F
Duplicate found:
        Boards: circuitplayground_express_displayio, circuitplayground_express, 
                circuitplayground_express_crickit
        VID:PID: 0x239A:0x8019
Duplicate found:
        Boards: metro_m4_express, cp32-m4
        VID:PID: 0x239A:0x8021
Duplicate found:
        Boards: pycubed, pycubed_mram
        VID:PID: 0x04D8:0xEC44
Duplicate found:
        Boards: feather_m0_adalogger, feather_m0_rfm9x, feather_m0_basic, feather_m0_rfm69
        VID:PID: 0x239A:0x8015
Duplicate found:
        Boards: sparkfun_qwiic_micro_no_flash, sparkfun_qwiic_micro_with_flash
        VID:PID: 0x1B4F:0x8D24
Duplicate found:
        Boards: pygamer_advance, pygamer
        VID:PID: 0x239A:0x803E
```

sommersoft, 2020-05-27:
Or, are you looking for more than just duplicates?

tannewt, 2020-05-27:
That's what I was thinking.

Looks like we'll need a whitelist as well. Some of them look legit (such as pycubed and pycubed_mram.) Others will need to be fixed.

PR #2966 (2020-09-17): Automate USB VID/PID Uniqueness Check
Fixes #2962 

I added in my assumptions for ignore-able, whitelisted boards.

I imagine this will sit open a while, during any VID/PID reassignment. I'm more than happy to update `mpconfigboard.mk` entries here, or just defer it all to merge in changes prior to merging this one.

Example Actions run: https://github.com/sommersoft/circuitpython/runs/714747984?check_suite_focus=true#step:22:1
_(Or...the example run in this PR will work too. Not sure what I was thinking... lol)_
closed 2020-09-17 by tannewt