adafruit/circuitpython #2969: Optimize the CI builds

k0d, 2020-05-28:
I'm opening this issue so that we can discuss how we speed up the builds. Here are a few ideas.

Docker image containing dependencies
Clean gcc-arm-toolchain folder so it contains only needed files (my initial tests show ~600mb can shrink to ~100mb).
Use GitHub Actions caching (https://help.github.com/en/actions/configuring-and-managing-workflows/caching-dependencies-to-speed-up-workflows#using-the-cache-action)

MicroDev1, 2023-02-17:
We now implement caching in most places and I don't think that there is a need for using a `docker` image in CI now.
We can still however publish a `docker` image with all the CP deps as a package.

tannewt, 2023-02-17:
Thanks @MicroDev1. I'll close this.

closed 2023-02-17 by tannewt