adafruit/circuitpython #2973: _bleio.ScanEntry.matches() is too strict for "all=True" matches

dhalbert, 2020-05-28:
`bleio_scanentry_data_matches()` in `shared-module/_bleio/ScanEntry.c` returns `false` prematurely if the the first prefix doesn't match the first structure and `!any` is `true`. It should try all the structures.

kevinjwalters, 2020-06-02:
When fixing this it would be useful to keep an eye on any code which might fail on matching sub-classes of one or two `Advertisement`s. As mentioned in discord, I've had the _same_ code running on three boards which transit to each other and receive each other's broadcasts. Sometimes one goes "deaf" and doesn't pick up a particular sub-classed `Advertisement` for >15s which is really odd compared to its neighbour which receives them fine. The "deaf" CPB then suddenly picks up the next sub-class in the protocol fine. These all have single `prefix` matches on the default matching scheme.

kevinjwalters, 2020-06-02:
Actually, I'm starting to think it's my code that's getting out of sync with itself. Now I've got my own matching code I can also see more detail when it doesn't match and I have a very small sequence of different `Advertisement` types. I accept the current one and one after but not the one before...

PR #3036 (2020-06-16): Fix ScanEntry prefix matching for any=False 
Fixes #2973.

Tested with a heart rate monitor advertisement with this program:

```python

from _bleio import adapter, Address, ScanEntry

def p(x):
    print(tuple(hex(b) for b in x))

# Reverse the two prefixes to test both orders.
prefixes = b"\x07" b"\x03\x0D\x18\x0A\x18\x25\x17"       b"\x03" b"\x19\x41\x03"
p(prefixes)
HRM_ADDR = Address(bytes(reversed(b"\xEC\x5D\x34\x85\x99\x11")), Address.RANDOM_STATIC)
# active=False: skip scan responses
for se in adapter.start_scan(active=False):
    #print(se.address, se.address.type, HRM_ADDR, HRM_ADDR. [... cut]
closed 2020-06-16 by tannewt