adafruit/circuitpython #2984: ulab min/max are broken on 5.3.0 with int16

kevinjwalters, 2020-05-31:
There's something odd going on with `ulab.numerical.min` and `ulab.numerical.max` with the int16 data type. Example below from **nRF52840**-based CLUE:

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>> import ulab
>>> import ulab.numerical
>>> length=6
>>> someint16s = ulab.array([0, 12345, 23456, -20_000, -30_000, 0], dtype=ulab.int16)
>>> ulab.numerical.min(someint16s)
0
>>> ulab.numerical.max(someint16s)
-20000
```

I'd expect `-30000` and `23456` out of those. Converting to `float` type:

```
>>> somefloats = ulab.array(someint16s)
>>> type(somefloats)
<class 'ndarray'>
>>> somefloats
array([0.0, 12345.0, 23456.0, -20000.0, -30000.0, 0.0], dtype=float)
>>> ulab.numerical.min(somefloats)
-30000.0
>>> ulab.numerical.max(somefloats)
23456.0
```

And everything's fine.

jepler, 2020-05-31:
@v923z can you take a look?

kevinjwalters, 2020-05-31:
Oh, I've not checked this on MicroPython. Perhaps this will end up in https://github.com/v923z/micropython-ulab - I hadn't thought about that.

v923z, 2020-06-01:
> @v923z can you take a look?

Yes. This is certainly odd, and should be fixed. Thanks for reporting!

v923z, 2020-06-01:
@jepler This was a trivial error that was entirely my mistake: in certain cases, you can treat uint16/int16 on the same footing, but not here. 

In any case, I have patched the code. https://github.com/v923z/micropython-ulab/pull/121 should fix the issue. If you agree, you can merge it into master. When pulling code into `circuitpython`, keep in mind that master contains the `approx` sub-module now, which you might not want to include on your end, so you have to unset the switch in `ulab.h`. 

@kevinjwalters Kevin, thanks for bringing this to the fore!

jepler, 2020-06-02:
Should be fixed since #2992, please reopen and comment if not.

closed 2020-06-02 by jepler