adafruit/circuitpython #2993: Have important (serial) things log to SD card

ztynzo, 2020-06-01:
It would be nice if anything that would normally go to serial could be set up to log automatically to the SD card.

For example, I've got a PyPortal, and really spotty internet. It seems to get hung up once a day, except when I leave a serial terminal watching it. If I know where and why it was hanging I'd be able to code around those issue to make it more reliable. (The Weather Station code from the PyPortal Adabox.)

If we could have anything that would normally go to serial also go to SD card when specified in code and card detect pin is set that would really help catch infrequent issues. Would probably only be needed on devices that ship with an SD card reader on them.

tannewt, 2020-06-01:
@jepler Food for thought as you do SD card work. This is a good, long-term idea.

ajs256, 2021-03-28:
Major +1 to this. It would be really handy for projects that aren't plugged into a computer all the time.