adafruit/circuitpython #2994: Enabling high brownout level too soon causes BOD reset with slow ramping supply

xorbit, 2020-06-01:
The brownout level is set quite high (2.77V) too soon after POR release on fast chips with slow ramping supplies.

An issue was encountered when testing the [PoE-FeatherWing](https://hackaday.io/project/168356-poe-featherwing) with the Adafruit Feather M4 Express.  It would continuously go into safe mode right after boot when connected to PoE.  My theory is that as the PoE power is still ramping up, the chip comes out of POR and starts to execute code.  It reaches `port_init` before the input voltage has ramped up enough for the regulated voltage to reach 2.77V when the [BOD is set to this level and turned back on](https://github.com/adafruit/circuitpython/blob/bdbb16ec8337eec0fa7c61d91b0494519d34d318/ports/atmel-samd/supervisor/port.c#L211), resulting in an immediate BOD reset.  The input voltage continues to ramp up so after running through the init code again it passes BOD successfully, but it is now running in "safe mode" and not executing any loaded code.

I verified that the issue goes away if I release the EN after a delay, meaning, I'm delaying the BOD detection level change long enough to where 3.3V has reached a high enough level by the time this is done, and the code runs.

While I can solve this with a hardware change (adding a 4.7uF cap from EN to GND seems to fix it), I feel like this is a more fundamental issue that deserves reporting and should be addressed.  There is a large gap between the ~1.6V POR level and the 2.77V BOR level, a lot of code can be executed in this time while the voltage ramps up.  This is only going to get worse with faster chips in the future.

Now I understand that you need this BOD level to make sure that chips that need external SPI flash to run are running in a supported voltage range.  I do not fundamentally have a problem with the BOD happening, but it is a problem that after it happens, the system goes into safe more and does not run any code.

I can see some possible ways to fix this that preserve the BOD's function:

- Add a delay at boot to give supplies time to settle before enabling the high BOD level.
- Better would be to wait a fixed amount of time before the BOD is set to the high level.  This makes it independent of the CPU speed, so things keep working if a user swaps one Feather for a future faster model for instance.
- Best option: you know whether initialization is happening from POR or from BOD.  Might it be possible to add a `finished_ init` flag that is not initialized by C, that is cleared on POR and set after a successful `port_init`.  When a BOD reset is detected, only go into safe mode if the `finished_init` flag is set.  That way you can tell the difference between "BOD reset happened during startup while the input voltage was still ramping up" and "BOD reset happened after the system initialized and we were running, so we should go to safe mode".

I think this is worth adding in general not just for my use case with the PoE-FeatherWing, but it would also help other use cases.  Ironically, this issue might bite someone who needs a very stable supply when switching high loads, so they have a lot of bulk capacitance.  Such a system would never see a real brown-out, but might not be able to start with the current BOD implementation.

deshipu, 2020-06-02:
I had a lot of problems with brownout safe mode recently too, I started wondering if something in the recent changes with timers made it much easier to get into safe mode.

dhalbert, 2020-06-02:
What version of the bootloader are you using? I added code starting a v3.9.0 that waits until the voltage reaches the BOD33 setting, and then waits 100 msecs: https://github.com/adafruit/uf2-samdx1/blob/master/src/main.c#L216

xorbit, 2020-06-02:
Excellent!  I had updated my 2 boards to the latest CircuitPython, but had not updated the bootloader.  One board had bootloader version 2.0.0 and the other 3.7.0.  I updated both and they now work correctly.

I will just have to tell my PoE-FeatherWing customers they need to update their bootloader to the latest version if theirs is below 3.9.0.

I closed the issue.

closed 2020-06-02 by xorbit

dhalbert, 2020-06-02:
Glad it worked for you! That's a good test.