adafruit/circuitpython #2998: teensy 4.1 and 4.0 builds not booting

jerryneedell, 2020-06-02:
The most recent builds for the teensy 4.1 and teensy 4.0 do nto boot for me
this fails to boot
```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/adafruit_github/teensy_loader_cli$ ./teensy_loader_cli -mmcu=imxrt1062 -w ~/Downloads/adafruit-circuitpython-teensy41-en_US-20200602-e889287.hex
```
this boost OK 
```
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/adafruit_github/teensy_loader_cli$ ./teensy_loader_cli -mmcu=imxrt1062 -w ~/Downloads/adafruit-circuitpython-teensy41-en_US-5.4.0-beta.0.hex 
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/adafruit_github/teensy_loader_cli$ 
```

same for teensy 4.0

jerryneedell, 2020-06-02:
the last build that works was:

```
adafruit-circuitpython-teensy41-en_US-20200519-916ca9f.hex 
```
it starts failing with 
```
adafruit-circuitpython-teensy41-en_US-20200520-0af5dd5.hex

tannewt, 2020-06-02:
@hierophect Do you have teensy's you can test this with?

hierophect, 2020-06-02:
Well, I've tried this on the 4.0 and something critical is getting reset outside the flash pins, I think. Commenting out reset_all_pins from port_reset results in normal operation so it's definitely a pin overwrite issue... but which pin?

tannewt, 2020-06-03:
> but which pin?

I'd suggest binary searching it. Make reset_all_pins only reset half of them and then repeat for the bad half.

hierophect, 2020-06-03:
@tannewt yeah I tried that this morning, it actually fails with *every* pin in never_reset. From what I could briefly gather with the Jlink before it stopped connecting altogether, the chip might actually be crashing when it hits the never reset portion?

hierophect, 2020-06-03:
True... I'll try stepping through the reset portion more carefully and try and figure out what's going on. Maybe it isn't adding them to never reset, somehow? It'd be weird, since it works just fine on the 1010

PR #3001 (2020-06-05): mimxrt10xx: Disable pin_reset on 1060 boards
This PR removes reset_all_pins for the 1060 line, as a temporary fix for what appears to be a general hard crash when pins in this line are reset. Also expands the never reset pin set in board.c to appropriately cover all flash chips on the Teensy 4.0 and 4.1, but since these pins will not be called by reset_all, this change will only affect future fixed commits.

This PR should be followed up by a more general fix for reset_all_pins on the 1060.

Resolves #2998
closed 2020-06-05 by tannewt