adafruit/circuitpython #2999: Merge in upstream MicroPython changes

tannewt, 2020-06-02:


theacodes, 2020-07-06:
A small bump on this, as it's something I'd really love to see happen.

Notable things that we'd get:

- Type annotations (PEP 526).
- Walrus assignment operator (PEP 572).
- Matrix multiplication `@` operator (PEP 465).
- Underscored numeric literals (PEP 515).
- Support for dynamically loading native modules in `mpy` files.
- Numerous code size improvements.
- Significantly better native code emitter.

I don't presently have the time (or expertise) to do this myself, but I would be interested in sponsoring this.

dhalbert, 2020-07-08:
I am willing to do this when I have time. We are thinking of waiting for the MicroPython 1.13 release.

benetherington, 2020-11-11:
Bump from me too!

MicroPython 1.13 was released in September, and the walrus assignment operator is very handy for tidy code.

timonsku, 2021-02-14:
+1 on this from me, being able to write native modules without needing to integrate with the core would be huge.

timonsku, 2021-02-15:
See https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules for a use case for this. This probably ties also into https://github.com/adafruit/circuitpython/issues/2825

deshipu, 2021-02-15:
@timonsku having been through the process of trying to use that, I have to say it's not as useful as it seems. The problem is that all the functions you are going to call have to be included in a special table at compile time — and only a handful is there by default — so while it's useful for speeding up some purely computational functions, it's very limited otherwise.

tannewt, 2021-02-16:
I'm also very concerned about the support burden of native modules that work for ARM but not xtensa or risc-v for example. I'd encourage folks to write native modules in the core instead. We're happy to have them.

garethhcoleman, 2021-02-27:
Is this separate issue still valuable alongside #4280?

tannewt, 2021-03-02:
> Is this separate issue still valuable alongside #4280?

Ya, I think so. Issues can live when inactive and PRs should be closed when they are. We can close this when the 1.14 merge is done.

closed 2021-05-13 by jepler

ladyada, 2021-05-13:
nice!

PR #4749 (2021-05-13): Merge in MicroPython 1.15
Fixes #2999 for now.