adafruit/circuitpython #3008: _bleio.ConnectionError conflicts with standard exception ConnectionError

dhalbert, 2020-06-04:
pylint will complain insome cases about `_bleio.ConnectionError` shadowing the original global `ConnectionError` exception.

Let's consider changing the name of the `_bleio` exception.

dhalbert, 2020-07-27:
@tannewt What do you think about adding a new name aliased to the old name for 6.0.0, and then drop in 7.0.0? I'm not sure what a better name is, maybe `ConnError` or `BluetoothConnectionError` or `BTConnectionError`.

dhalbert, 2020-07-28:
The basic problem is that:
```python
from _bleio import ConnectionError
```
will trigger a pylint complaint, because it's shadowing the system exception.

Another alternative I thought of is simply to drop `_bleio.ConnectionError`, and use the system-provided `ConnectionError` for that particular error. We don't provide `_bleio.IOError`, etc., so using the system `ConnectionError` may be clear enough.

tannewt, 2020-07-28:
Using the native one is fine with me. You could have the BLE version inherit from ConnectionError to start.

PR #3523 (2020-10-08): Replace _bleio.ConnectionError with the native version
Replace uses of _bleio.ConnectionError with regular ConnectionError

Fixes #3008
commit 9fcf96cb (2020-10-08): Replace _bleio.ConnectionError with the native version

Replace uses of _bleio.ConnectionError with regular ConnectionError

Fixes #3008
closed 2020-10-08 by tannewt