adafruit/circuitpython #3011: paste to REPL causes problems on MIMXRT boards

jerryneedell, 2020-06-05:
Just a heads up for now -- I'l try to characterize this better.

when I copy/paste code to the REPL on MIMXRT boards I  have seen the following
on Teensy 41 - trying to past this
```
import adafruit_sdcard
import digitalio
import board
import storage
import sys
# Connect to the card and mount the filesystem.
spi = board.SPI()
```
causes the board to enter the bootloader (red LED starts "breathing" - CIRCUITPY drive disconnects)

on the IMIXRT1010-EVK - a similar large-ish past causes the board to just hang - REPL remains "connected" but unresponsive. Control C does not work -- have to RESET to get it back.

some copy/past operations work fine on both boards. I have not come up with a cleal picture of what triggers the issue. I'll do more testing and report.

jerryneedell, 2020-06-05:
better example
on both the teensy 41 ans imxrt1010-evk
```
0123456789
0123456789
0123456789
0123456789
0123456789
```  work OK but adding one line
```
0123456789
0123456789
0123456789
0123456789
0123456789
0123456789
```
causes it to crash -- teensy4.1 enters bootloader - imixrt1010-evk  just hangs

hathach, 2020-06-05:
maybe related to 
- https://github.com/hathach/tinyusb/issues/423 
- https://github.com/micropython/micropython/issues/6034

hathach, 2020-07-29:
@jerryneedell I submitted an PR to fix this #3225 , when you have time, please check it out.

PR #3225 (2020-07-30): Improve highspeed usb
- fix #3011 
- update TinyUSB for better highspeed support
  - for Highspeed all bulk endpoint (cdc, midi, msc etc ...) must be **exactly** 512
  - add `--highspeed` option to replace --msc_max_packet_size
  - make sure all CFG BUFSIZE for usb is at least 512 for highspeed port ( iMXRT & cxd56 )
- Should also improve esp32s2 usb speed thanks to https://github.com/hathach/tinyusb/pull/468
- Also remove obsolete tinyusb config macros
closed 2020-07-30 by jepler