adafruit/circuitpython #3013: Requesting a USB VID/PID for nice!nano

Nicell, 2020-06-05:
I'm requesting a VID/PID for my new nRF52840 development board called the "nice!nano" that will be available soon. https://docs.nicekeyboards.com/#/nice!nano/ It's a general purpose board that many will be using for keyboards.

Thanks!

ladyada, 2020-06-05:
@nicell will you be submitting a circuitpython board support definition?

Nicell, 2020-06-05:
Yes! I was taking a look at the other nRF boards, and it looks like I should be able to add a definition quickly.

ladyada, 2020-06-05:
great :) 
```
nice!nano         VID 0x239A    PID 0x00B3  # bootloader
                                PID 0x80B3  # arduino
                                PID 0x80B4  # circuitpython
```

closed 2020-06-05 by ladyada