adafruit/circuitpython #3019: Can not get a lock on the I2C bus

geekguy-wy, 2020-06-07:
I have been running into a problem off and on where a script can not get a lock on the I2C bus. This is one of those soft failures that only shows up once in a while. :( I have a small script that demonstrates this problem when it occurs. I can scan the I2C bus successfully and show what devices are there, but I can not instantiate a display. I believe this _might_ be a bug in the underlying C code.
[i2cscan_displays.zip](https://github.com/adafruit/circuitpython/files/4743228/i2cscan_displays.zip)

jerryneedell, 2020-06-08:
in your example , you are locking the I2C bus -- I think you have to unlock before using it
try adding i2c.unlock() before you initialize the disply
```
#!/usr/bin/env python3
import board
import busio
from time import sleep

from adafruit_ht16k33.segments import Seg14x4

i2c = busio.I2C(board.SCL, board.SDA)

displays = []

print()
print("Simple I2C bus scanner")
print()

while not i2c.try_lock():
	sleep(0.5)

scan = i2c.scan()

if scan:
	print("Found:")

	for addr in scan:
		print("\tDecimal: {0:3d}, Hex, {1}".format(addr, hex(addr)))
                i2c.unlock()
		print("\tAttempting to initialize the display at address {0}".format(hex(addr)))
		displays.append(Seg14x4(i2c, addr))
else:
	print("There are no devices on the I2C bus")

```
this worked for me

geekguy-wy, 2020-06-08:
Well, this does seem to work! I moved the unlock to the beginning of the if block and it still does work, although I still see an occasional lock up. I have never had to do this before but will start doing it now. I will close this issue for now and reopen it if necessary.

Thank you!

closed 2020-06-08 by geekguy-wy