adafruit/circuitpython #3021: Rename "master" branch to "main"

tannewt, 2020-06-08:
It's a clearer term with less history.

dhalbert, 2020-06-08:
https://github.com/desktop/desktop/issues/6478 did something equivalent, and mentions a few things to make sure to change. Also, there are mentions of that issue by a bunch of other folks, so check out the linked issues at the end, for further things  to check for.

tannewt, 2020-06-08:
We could also rename ***all*** of our repos programmatically.

tannewt, 2020-06-09:
Here is a blog with tips from @shanselman : https://www.hanselman.com/blog/EasilyRenameYourGitDefaultBranchFromMasterToMain.aspx

tannewt, 2020-06-09:
Had one github action to update: #3024

jerryneedell, 2020-06-12:
FYI -- just came across this https://www.washingtonpost.com/opinions/2020/06/12/tech-industry-has-an-ugly-master-slave-problem/?utm_campaign=wp_opinions_pm&utm_medium=email&utm_source=newsletter&wpisrc=nl_popns

stefanoborini, 2020-06-15:
-1. Context matters. This is newspeak.

tannewt, 2020-06-22:
We want to do this in the long term but will likely wait to hear more from GitHub until we do all of the library repos. CircuitPython core is already switched over.

tannewt, 2021-08-03:
@kattni Have we finished switching libraries over?

kattni, 2021-08-03:
@tannewt For the most part. Everything else is being tracked within its own repo. This can be closed.

closed 2021-08-03 by kattni