adafruit/circuitpython #3022: More visible warning on old version docs pages

FoamyGuy, 2020-06-08:
On the older version docs pages such as this one: https://circuitpython.readthedocs.io/en/4.x/shared-bindings/displayio/Group.html

There is a warning at the top and center but I find it somewhat easy to overlook when I arrive on a docs page from a google search or other link.

The two ideas for improvement I had were to make the warning a bolder color and/or make it scroll with the page so it's always visible.

FoamyGuy, 2020-06-08:
I use the Django docs somewhat frequently as well and tend to overlook the warning less. Here is an example of their out of date pages: https://docs.djangoproject.com/en/2.1/

The warning is a top bar and it stays with the user as they scroll.

tannewt, 2020-06-09:
@sommersoft Any idea about how to implement this?

dhalbert, 2020-06-09:
This is not hard to do in CSS, something like:
.warning {
```css
  position: fixed; /* Set to fixed position */
  top: 0; /* Position at the top of the page */
}
```

sommersoft, 2020-06-09:
I actually tried that in developer mode (with `position: sticky`). Didn't work for me (yet). Also, there is no unique ID for that box; just using `admonition` or `warning` would apply to any of them in the documentation. Which, we have some scattered throughout.

That box is inserted by RTD's rendering, so without a good way to CSS it, there's not much to do. I peeked in the RTD docs, and it's barely mentioned that this happens, and there is nothing really beyond that. https://docs.readthedocs.io/en/stable/versions.html#version-warning

FoamyGuy, 2020-06-09:
I'm not sure I fully understand how Sphinx and Read the Docs work together to create the docs pages (or if there are other components at work). So perhaps I am overlooking something. It looks like there is support for custom js code: https://docs.readthedocs.io/en/stable/guides/adding-custom-css.html

It would be ideal if that warning already had a unique id or class to target with a CSS rule. But since it doesn't maybe it's possible to use custom JS code to find the version warning box based on it's contents and then add an id or class to it.

sommersoft, 2020-06-09:
I actually just found out why setting the `position` in CSS isn't working. RTD's theme sets `overflow-x: hidden` on the entire HTML body, which breaks the usage of `position`.

Disabling `overflow-x` allows it to sticky, but there are some side-effects (more than I'm aware of, I'm sure). 

I can at least select the only box we're worried about with the proper CSS selector:
```css
.document > .admonition {
    position: sticky;
    top: 0px;
}
```
One side-effect I'm seeing is some float-like behavior:
[image]

sommersoft, 2020-06-10:
This invocation produces fairly good results for me on Chrome:
```css
.document > .admonition {
    position: sticky;
    top: 0px;
    background-color: salmon;
    z-index: 2;
}

body {
    overflow-x: unset!important;
}
```
4.x and earlier don't seem to call the `docs/static/customstyle.css` on RTD, so that will need to be fixed for it to be applied I think.

[image]

FoamyGuy, 2020-06-12:
hierophect asked in the discord about a way to hide the notification if it is made sticky. I agree that we should try provide a way to do that if possible. 

Here is a bit of Javascript that will hide the notification when it gets clicked on:
```javascript
$(".document > .admonition").click(function(){$(this).hide()});
```
I tested in with the browser console on the live docs page and it seems to be working. 

Clicking on the latest version link does fire this listener and hide the box, but it does not interfere with being taken to the next page. If we wanted we could target a smaller zone for click-ability like just the title bar, or add an X in the corner.

simontorres, 2021-05-16:
So, this has been already solved? should this issue be closed?

tannewt, 2021-05-18:
Yup! Thank you @simontorres

closed 2021-05-18 by tannewt