adafruit/circuitpython #3029: External Interrupt

CaptainJackey, 2020-06-13:
Will circuitpython add the GPIO External Interrupt funtion later?

tannewt, 2020-06-15:
We won't be adding the external interrupt callback that MicroPython has.

At some point we'd like to support external interrupts as a sleep wakeup though.

closed 2020-06-15 by tannewt

CaptainJackey, 2020-06-16:
> We won't be adding the external interrupt callback that MicroPython has.
> 
> At some point we'd like to support external interrupts as a sleep wakeup though.

Imagine that scene, I want to use a GPIO input (SWITCH) to control several funtions of a system, and maybe some funtions have delay inside. And how could check the SWITCH efficiently without External Interrupt in CircuitPython?

tannewt, 2020-06-16:
See #1380