adafruit/circuitpython #3030: CircuitPython can't play / loop samples and BLE start_scan at the same time - memory corruption?

kevinjwalters, 2020-06-13:
I had some code which was working well until I changed the WaveFile use to include a pre-allocated 2048 `bytearray`. That new code fell over in different bizarre ways on three boards, one went into the hard fault handler.

I've experimented a bit with REPL and I believe one way of triggering this is with adafruit_ble's `start_scan()`. This is where the real application falls over too although it's interesting that it worked perfectly with WaveFile doing its own thing for buffers.

Here's a reproduction of the issue on the nRF52840-based Adafruit CLUE.

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>>
>>> import board
>>> from displayio import Group
>>> import displayio
>>> from audiocore import WaveFile
>>> from adafruit_clue import clue
>>> import gc
>>> gc.collect()
>>> print(gc.mem_free())
91616
>>> file_buf = bytearray(2048)
>>> from audiopwmio import PWMAudioOut as AudioOut
>>> audio_out = AudioOut(board.SPEAKER)
>>> audio_out.play(WaveFile(open("rps/audio/arena.wav", "rb"), file_buf), loop=True)
>>> from adafruit_ble import BLERadio
>>> ble = BLERadio()
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x112233
>>> ble.start_advertising(ca)
>>>
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Advertisement' is not defined
>>> from adafruit_ble.advertising import Advertisement
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
>>> for adv in ble.start_scan(Advertisement, minimum_rssi=-120, timeout=10):
...     print(adv)
...
...
...
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
<Advertisement tx_power=-10 flags=<AdvertisingFlags limited_discovery > >
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "adafruit_ble/advertising/__init__.py", line 290, in __str__
TypeError: can't convert 'float' object to str implicitly
>>> for adv in ble.start_scan(Advertisement, minimum_rssi=-120, timeout=10):
...     print(adv)
...
...
...
```

Nothing more appears on the serial connection as it's now kaput. The screen on the CLUE is useful here as it reports: `Running in safe mode: Not running saved code.`

I suspect that `TypeError: can't convert 'float' object to str implicitly` is bogus and just caused my memory corruption.

BTW, would be useful to have some tips about whether the bytearray size should vary based on the wav file rate and/or playback rate in the docs. The docs also mention 2x512 but I've seen `MemoryError` exceptions for 2048 bytes when I run out of memory.

dhalbert, 2020-06-14:
Something is clearly going wrong with storage management, so we'll need to look at the heap allocation that happens during scanning, and whether the internal audio buffers are being managed properly (they are on the root pointer list, etc.). @jepler does this ring any bells w/r/t audio?

Thanks for coming up with a really simple example that is just audio playing in the background during a scan. That narrows things down considerably, since there's no audio Python code that's being called in a loop or anything like that.

jepler, 2020-06-14:
I have also seen a hard fault using WaveFile(buffer=), and bluetooth wasn't involved.  @deshipu It looks like you contributed the buffer= code, is it working for you?

kevinjwalters, 2020-06-14:
BTW, it works in my code for a bit and even when there's `displayio` animations going on. There's some sort of combination effect here, possibly more heavy use of memory or GC'ing revealing a bug.

kevinjwalters, 2020-06-28:
This is just blew (LCD shows `safe mode`) up at the import of ble code so probably nothing to do with ble stuff. One difference here is I have refreshed libraries and I no longer have the `adafruit_clue`.

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>>
>>>
>>>
>>>
>>> import board
>>> from displayio import Group
>>> import displayio
>>> from audiocore import WaveFile
>>> from adafruit_clue import clue
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'adafruit_clue'
>>> import gc
>>> gc.collect()
>>> print(gc.mem_free())
144048
>>> file_buf = bytearray(2048)
>>> from audiopwmio import PWMAudioOut as AudioOut
>>> audio_out = AudioOut(board.SPEAKER)
>>> audio_out.play(WaveFile(open("rps/audio/arena.wav", "rb"), file_buf), loop=True)
>>> from adafruit_ble import BLERadio
Traceback (most recent call last):
```

kevinjwalters, 2020-06-28:
I don't know if these end up contiguous but it seemed worht a shot. Something clearly isn't healthy but there's no massive splat into `spare1` and `spare2`.

@jepler Do you know for your case if you were looping?

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import board
>>> import displayio
>>> from audiocore import WaveFile
>>> from adafruit_clue import clue
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'adafruit_clue'
>>> import gc
>>> gc.collect()
>>> print(gc.mem_free())
144080
>>> spare1 = bytearray(2048)
>>> file_buf = bytearray(2048)
>>> spare2 = bytearray(2048)
>>> from audiopwmio import PWMAudioOut as AudioOut
>>> audio_out = AudioOut(board.SPEAKER)
>>> audio_out.play(WaveFile(open("rps/audio/arena.wav", "rb"), file_buf), loop=True)
>>> from adafruit_ble import BLERadio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: byte code not implemented
>>> spare1
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x [... cut ...] 00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
>>>
```

kevinjwalters, 2020-06-28:
_Probably_ related to the buffer size, if I use 512 which is the equivalent of the what's allocated in the code (2*256) then it seems robust, admittedly form one quick test. This makes some sense as this is exercising less used/tested code, perhaps?

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>>
>>>
>>> import board
>>> from displayio import Group
>>> import displayio
>>> from audiocore import WaveFile
>>> import gc
>>> gc.collect()
>>> print(gc.mem_free())
144304
>>> file_buf = bytearray(512)  ### this looks like equivalent size of default allocation
>>> from audiopwmio import PWMAudioOut as AudioOut
>>> audio_out = AudioOut(board.SPEAKER)
>>> audio_out.play(WaveFile(open("rps/audio/arena.wav", "rb"), file_buf), loop=True)
>>> from adafruit_ble import BLERadio
>>> ble = BLERadio()
>>> from adafruit_ble.advertising.adafruit import AdafruitColor
>>> ca = AdafruitColor()
>>> ca.color = 0x445566
>>> ble.start_advertising(ca)
>>>
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Advertisement' is not defined
>>> from adafruit_ble.advertising import Advertisement
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
<Advertisement tx_power=12 flags=<AdvertisingFlags general_discovery > >
>>> for adv in ble.start_scan(Advertisement, timeout=20):
...     print(adv)
...
...
...
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement  >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement  >
<Advertisement  >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement tx_power=0 complete_name=CIRCUITPYxxxx >
<Advertisement  >
<Advertisement  >
<Advertisement  >
>>>
```

kevinjwalters, 2020-06-29:
Using a `bytearray(512)` is good in my application too which plays a lot of samples including a few looping ones.

jepler, 2020-06-29:
@kevinjwalters Is use of the second argument of WaveFile (the optional buffer) required in order to trigger this problem?

kevinjwalters, 2020-06-29:
@jepler Yes, second arg makes it go haywire from what I have observed. But passing in a `bytearray(512)` (note length) is ok so far in my application...

kevinjwalters, 2020-07-10:
This looks a bit fishy:

1. https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c#L268-L279
2. https://github.com/adafruit/circuitpython/blob/main/ports/nrf/common-hal/audiopwmio/PWMAudioOut.c#L252-L265
3. https://github.com/adafruit/circuitpython/blob/main/ports/nrf/common-hal/audiopwmio/PWMAudioOut.c#L98-L102
4. https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c#L186-L209

Also, is `audioio_wavefile_max_buffer_length()` from https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c#L167-L169 used anywhere in the code? That has an independent copy of the `512` value from https://github.com/adafruit/circuitpython/blob/main/shared-module/audiocore/WaveFile.c#L273