adafruit/circuitpython #3032: Weird markup problem in autoapi docs

jepler, 2020-06-14:
I noticed that this bit of documentation in shared-bindings/displayio/Palette.c:
```
//|           palette[1] = b'\xff\xff\x00'              # set using 3 bytes
```
gets rendered as
```
palette[1] = b'ÿÿ'              # set using 3 bytes
```
i.e., the hex escapes are being interpreted.  It appears that this happens at the step where the .pyi file is converted to a .rst file, as the generated files have these contents:
```
autoapi/displayio/index.rst:        palette[1] = b'ÿÿ'              # set using 3 bytes
circuitpython-stubs/displayio/__init__.pyi:          palette[1] = b'\xff\xff\x00'              # set using 3 bytes
```

This may be an autoapi bug that we should report upstream?

dhalbert, 2020-06-14:
Definitely we should report it, I think here: https://github.com/readthedocs/sphinx-autoapi/issues. I looked and didn't immediately see an existing issue.

jepler, 2020-06-14:
On reflection, I think this may be "correct", and there's an easy fix.  I've incorporated it in #3031 where I was already working on the docs, but I can split it out if desired.

If you write
```
"""This is a byte string with hex escapes: b'\x00\xff'"""
```
then the string will actually have those characters with code points 0 and 0xff in it.  By changing these docstrings to _raw_ strings we get what we want.

dhalbert, 2020-06-14:
Ah, I see, a docstring is just a string, subject to escape processing. The alternative would be to use double-blackslashes.

commit 32a29ffd (2020-06-16): shared-bindings: Change docstrings with '\x...' chars to raw strings

Closes: #3032
closed 2020-06-16 by tannewt