adafruit/circuitpython #3049: Pulseio not  working properly on M0

jerryneedell, 2020-06-19:
see issue for DHT sensor 
https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/45

something in the M0 build broke with pulseio between 5.3.0 and 5.4.0-beta1
at least - DHT driver no longer works

I seems to still be OK on the M4

DavePutz, 2020-06-26:
Just as another data point; using adafruit_irremote on a CircuitPlayground Express (as the example at https://learn.adafruit.com/remote-control-tree-ornament-with-circuit-playground-express/code-the-circuit-playground-express shows) experiences the same issues. It works using CP 5.3, but fails using the current beta 5.4.0-beta1.
It seems like a timing issue introduced in 5.4. @tannewt , is there anything I could look at or try here?

tannewt, 2020-06-26:
It's likely an issue with the new reference clock. ticks_ms doesn't run anymore so a separate timer is needed for higher resolution. It'd be great if you could double check my math on it. I may have also messed up the interrupt. Not exactly sure. Thanks!

DavePutz, 2020-07-05:
The issue has been traced to the amount of time it takes to synchronize the RTC timer when the COUNT field is accessed. It can take more than 100us, which exceeds the length of the incoming DHT pulses. As a result, interrupts will be missed. The solution is to utilize the continuous synchronization ability of the RTC. Since that setting will be reset anytime an RTC register is written, those writes , which are primarily done in port_interrupt_after_ticks(), will need to be avoided for the duration of the pulsein interrupts. I will put together a PR to implement this functionality.

jerryneedell, 2020-07-08:
Fixed by #3120

closed 2020-07-08 by jerryneedell