adafruit/circuitpython #3053: Tab completion invokes property

dhalbert, 2020-06-19:
Hitting tab to auto-complete a property name invokes the property.

testtab.py:
```python
class C:
    @property
    def oops(self):
        print("OOPS!")
```

in REPL:
```python
Adafruit CircuitPython 5.4.0-beta.1-37-g2b652e1d1-dirty on 2020-06-19; Adafruit Metro M4 Express with samd51j19
>>> import testtab
>>> c = testtab.C()
>>> c.oops
OOPS!
>>> c.oo<TAB>OOPS!     # <TAB> is where I hit the tab key
ps
```

siddacious, 2020-06-19:
Yes! Please! I'm sorry I didn't think to report this because it bites me _regularly_

jepler, 2020-06-19:
cpython 3 does this as well.
```
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import oops
>>> c = oops.C()
>>> c.oo<tab>OOPS!
ps
```

dhalbert, 2020-06-19:
It's not a bug, it's a feature. Oy. :facepalm: 
https://bugs.python.org/issue29630 (_REPL tab-completion triggers code execution_). Closed because "that's the way it works".

closed 2020-06-19 by dhalbert