adafruit/circuitpython #3058: IMIXRT1010-EVK excessive RAM usage

PTS93, 2020-06-22:
I was trying out [this example](https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/master/examples/st7789_320x240_simpletest.py) for the ST7789 which caused a crash due to out of RAM error it needs about 2kb more RAM than it can allocate.
`gc.mem_free()` returns 26160 bytes being available before importing any module. Given that there is 64kb available RAM that already seems awfully small.

After importing the modules it drops to 17664 bytes.
And after creating the first bitmap at [this point](https://github.com/adafruit/Adafruit_CircuitPython_ST7789/blob/master/examples/st7789_320x240_simpletest.py#L34) only 5824 bytes are left.

So something feels off here, especially about the low initial amount of RAM given that there is another 64kb RAM that is apparently used for cache?
On a Feather M4 with 192kb RAM
The RAM usage in the same order is:
157376
151840
139952

so at least the modules construction needs quite a bit more RAM.

tannewt, 2020-06-22:
64k of ram is used for the instruction tightly coupled memory (TCM) and the data TCM. This is done so that code can execute as fast as possible.

I'm not sure why the starting heap is so small. It may be accounting for the stack even though the stack should be placed in the TCM.

We don't have cycles to look into the iMX RT bugs at the moment so let me know if you need help looking further into it.

PTS93, 2020-06-23:
I think that issue might go a bit too deep for my current level of understanding of the CPY core :)
I'm just evaluating the i.MX RT range to see if they work out for a project. If work on the i.MX RT is on hold atm then thats good to know.

tannewt, 2020-06-23:
We're more focused on the ESP32-S2 at the moment since they've become available.

I'm happy to help you dive into CircuitPython if you like. :-)

PTS93, 2020-06-23:
I figured that focus shifted there :)
An intro would be nice actually. I'll ping you on Discord.

tannewt, 2023-03-21:
We now have just over 40k free:
```
Adafruit CircuitPython 8.1.0-beta.0-45-g461d833c1-dirty on 2023-03-21; Metro MIMXRT1011 with IMXRT1011DAE5A
>>> import gc
>>> gc.mem_free()
42864
```

I reduced some of the USB memory buffers to make more space. There is still about 4k of USB buffers and 4k for the flash buffer plus other static variables.

closed 2023-03-21 by tannewt