adafruit/circuitpython #3059: STM32: Move NVM into filesystem region

hierophect, 2020-06-22:
Currently, enabling the NVM module for STM32 requires an entire sector of memory, which is at minimum 16kB on the STM32. This section should be reworked to use a small amount of memory after the filesystem through the existing internal flash manager, rather than an entire linker-defined sector.

dhalbert, 2020-06-22:
If the filesystem and NVM share a single erase sector, then a write to either the filesystem or NVM will erase the page temporarily and will cause data in the other functionality to be wrong temporarily. So the writes will have to be atomic from the other functionality's point of view.

hierophect, 2020-06-22:
@dhalbert the filesystem is cached, but I guess NVM cannot be since the user counts on anything they write being saved right away. So I think the implementation would need to flush the filesystem cache every time NVM writes, but not vice versa.

dhalbert, 2020-06-22:
The scenario I am thinking of is:
1. User writes to nvm
2. page erase happens
3. host computer fetches filesystem data; this must be locked out until 4.
4. nvm write completes

hierophect, 2020-06-22:
@dhalbert yeah that's why I suggested putting it in the internal_flash module, it should just be alongside the rest of the flash operations where everything is guaranteed to be sequential.